LibreDWG/libredwg #439: Failed to decode

FishOrBear, 2022-04-22:
```
D:\emsdk\libredwg\test\test-data\2018>dwg2dxf taihe.dwg taihe.dxf
Reading DWG file taihe.dwg
Warning: checksum: 0x32761371 (calculated) mismatch

Warning: Invalid section_array_size: [8827].792575 != 8830
Warning: Invalid last_section_address: 18148ce != 1771280
Warning: section_map_id 8830 not found
Warning: Add section_map_id [8828] 8830 => address 0x174a220
Warning: Repair invalid section_info_id [8827]: => 8829
Warning: Add section_info_id [8829] 8829 => address 0x101680152
ERROR: Invalid Data Section Page Map type 0x97f73a7f != 0x4163003b
ERROR: Failed to decode file: taihe.dwg 0x100

READ ERROR 0x100

Usage: dwg2dxf [-v[N]] [--as rNNNN] [-m|--minimal] [-b|--binary] DWGFILES...

D:\emsdk\libredwg\test\test-data\2018>
```
[taihe.zip](https://github.com/LibreDWG/libredwg/files/8536188/taihe.zip)

rurban, 2022-04-22:
The Section Page Map is already corrupt. Wrong crc, and then a fatal error at Section[5889]. No idea, sorry

FishOrBear, 2022-04-22:
yes acad can be opened

rurban, 2022-04-23:
So it's an extremely good testcase. Thanks a lot!

rurban, 2025-03-16:
Can be read and converted with ACadSharp, so I can work on fixes

rurban, 2025-03-24:
I'm rewriting the decompressor and compressor to work with our streams API (Bit_Chain), and helpers as documented. Works fine with ACadSharp

rurban, 2025-03-26:
Fixed it, but doing more refactoring for read_R2004_section_info() right now

rurban, 2025-04-17:
Fixed now the 2004 decompressor

closed 2025-04-19 by rurban