LibreDWG/libredwg #444: Issue with block table parsing in preR13

michal-josef-spacek, 2022-04-24:
There is issue in block table parsing in preR13
Example: 
[BLOCK1.DXF.gz](https://github.com/LibreDWG/libredwg/files/8549851/BLOCK1.DXF.gz)
[BLOCK1.DWG.gz](https://github.com/LibreDWG/libredwg/files/8549852/BLOCK1.DWG.gz)

Actual parsing:
```
...
blocks   0x46c (0xfffffb94) - 0x0 (0x48, 0x80000000)
...
```
and code:
```
error |= decode_preR13_entities (blocks_start, blocks_end,
                                 blocks_start - blocks_offset, dat, dwg);
```
This doesn't work because block_end is 0. We need to use block_start + modified block_offset (& 0x00ffffff)

rurban, 2022-04-26:
Yes, fixed in 16f6c1f97b20fd8fdec1c595e24d2fa81801afce

michal-josef-spacek, 2022-04-26:
@rurban Thanks. I will test.
FYI: Mask works for me.

rurban, 2022-04-26:
yes, I re-introduced the mask, but renamed the field to size again. it's not an offset.

michal-josef-spacek, 2022-04-26:
@rurban great

michal-josef-spacek, 2022-04-26:
@rurban This change doesn't work.
You added num_entities to decode_preR13_entities(), but we don't know this value from R_10. In <R_10 is in header.
I believe that only thing how to do is while() and check entities_end value.

michal-josef-spacek, 2022-04-27:
Something like this: https://github.com/LibreDWG/libredwg/pull/452

michal-josef-spacek, 2023-01-04:
Fix for this issue: https://github.com/LibreDWG/libredwg/pull/559

closed 2023-01-16 by rurban