LibreDWG/libredwg #446: SVG file can't be recognized

laien1900, 2022-04-25:
The transformed SVG  file by dwg2SVG.exe is not recognized by browser

realiti4, 2022-04-25:
I am facing a similar issue. I am not sure what I am doing wrong.

rurban, 2022-04-26:
Sorry, but without any reproducer or diagnostics nobody can help you

laien1900, 2022-04-26:
```
	<path id="dwg-object-1073" d="M 546557.818098,-581161.941138 L 546557.818098,-581161.941138 L 546557.818098,-581161.941138 L 546557.818098,-581161.941138 L 546557.818098,-581161.941138 L 546557.818098,-581161.941138 L 546557.818098,-581161.941138 L 546557.818098,-581161.941138 L 546557.818098,-581161.941138 L 546557.818098,-581161.941138 L 546557.818098,-581161.941138 L 546557.818098,-581161.941138"
	      style="fill:none;stroke:black;stroke-width:0.0px" />
```
All the points above are the same value.

michal-josef-spacek, 2022-04-26:
@laien1900 Could you upload example dwg and output svg? (if will be simple, will be better).

laien1900, 2022-04-26:
> @laien1900 Could you upload example dwg and output svg? (if will be simple, will be better).
DWG file upload is not supported, so you need rename jpg as dwg.
[image]
[image]

michal-josef-spacek, 2022-04-26:
@laien1900 I couldn't read attachment. The best way is gzip/zip files and upload to issue.

michal-josef-spacek, 2022-04-26:
@laien1900 ah, i have files (from email notification).

rurban, 2023-02-10:
Either a duplicate of #523 (wrong viewbox) or missing support for curves and splines #520

closed 2023-02-10 by rurban