LibreDWG/libredwg #447: Duplicated parsing of tables in preR13

michal-josef-spacek, 2022-04-26:
There are two same blocks for parsing of tables in preR13
src/decode.c
e.g. BLOCK table: https://github.com/LibreDWG/libredwg/blob/master/src/decode.c#L389
```
...
    case SECTION_BLOCK:
      for (i = 0; i < tbl->number; i++)
        {
          PREP_TABLE (BLOCK_HEADER);
          FIELD_RC (flag, 70);
          FIELD_TFv (name, 32, 2);
          FIELD_RSd (used, 0); // -1
          FIELD_RC (block_scaling, 0);
          FIELD_CAST (num_owned, RS, BL, 0);
          FIELD_RCd (flag2, 0);
          FIELD_CAST (num_inserts, RS, RL, 0);
          FIELD_RSd (insert_units, 0);
          CHK_ENDPOS;
        }
      break;
...
```

src/dwg.spec
e.g. BLOCK table: https://github.com/LibreDWG/libredwg/blob/master/src/dwg.spec#L3066
```
...
 PRE (R_13)
  {
    FIELD_RC (flag, 70);
    FIELD_TFv (name, 32, 2);
    FIELD_RSd (used, 0); // -1
    FIELD_RC (block_scaling, 0);
    FIELD_CAST (num_owned, RS, BL, 0);
    FIELD_RCd (flag2, 0);
    FIELD_CAST (num_inserts, RS, RL, 0);
    FIELD_RSd (insert_units, 0);

    FIELD_VALUE (anonymous)    = FIELD_VALUE (flag) & 1;
    FIELD_VALUE (hasattrs)     = FIELD_VALUE (flag) & 2;
    FIELD_VALUE (blkisxref)    = FIELD_VALUE (flag) & 4;
    FIELD_VALUE (xrefoverlaid) = FIELD_VALUE (flag) & 8;
  }
...
```

I think we need to cleanup.

rurban, 2022-04-26:
yes, but later, not now.
I just need to put PREP_TABLE into dec_macros.h
and the end check also.

michal-josef-spacek, 2022-04-26:
@rurban I have patch to fix this decoding. Could i change both versions?

rurban, 2022-04-26:
Yes, so far we have to patch both versions. esp VIEW is broken

closed 2022-04-30 by rurban