LibreDWG/libredwg #449: Compilation issue

michal-josef-spacek, 2022-04-26:
master branch (e.g. https://github.com/LibreDWG/libredwg/commit/59c0694f25bde5294869fc2a1be620f0abb768e4)
```
./configure --disable-bindings --enable-write --enable-trace --disable-dxf --disable-json --disable-python --enable-debug
```
make
Output:
```
...
/usr/bin/ld: ../src/.libs/libredwg.so: undefined reference to `in_postprocess_handles'
/usr/bin/ld: ../src/.libs/libredwg.so: undefined reference to `dwg_encrypt_SAT1'
/usr/bin/ld: ../src/.libs/libredwg.so: undefined reference to `in_postprocess_SEQEND'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:716: dwgread] Chyba 1
make[2]: Opouští se adresář „/home/skim/44_git/libredwg/programs“
make[1]: *** [Makefile:678: all-recursive] Chyba 1
make[1]: Opouští se adresář „/home/skim/44_git/libredwg“
make: *** [Makefile:561: all] Chyba 2
```

rurban, 2022-04-30:
But only in the work/preR13 branch, right?

michal-josef-spacek, 2022-04-30:
@rurban Actual master has this issue

closed 2022-04-30 by rurban

michal-josef-spacek, 2022-04-30:
@rurban Great, thank you.