LibreDWG/libredwg #456: Error while make

overide, 2022-05-02:
Hi,

I'm following README for installation and was getting stuck while `make` in the EC2 machine. I was able to successfully install it in my local machine though - 

```
make  all-recursive
make[1]: Entering directory '/home/ubuntu/libredwg'
Making all in src
make[2]: Entering directory '/home/ubuntu/libredwg/src'
(CDPATH="${ZSH_VERSION+.}:" && cd .. && /bin/bash /home/ubuntu/libredwg/build-aux/missing autoheader)
rm -f stamp-h1
touch config.h.in
cd .. && /bin/bash ./config.status src/config.h
config.status: creating src/config.h
config.status: src/config.h is unchanged
make  all-am
make[3]: Entering directory '/home/ubuntu/libredwg/src'
  CC       dwg.lo
  CC       common.lo
  CC       bits.lo
  CC       decode.lo
gcc: internal compiler error: Killed (program cc1)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:648: recipe for target 'decode.lo' failed
make[3]: *** [decode.lo] Error 1
make[3]: Leaving directory '/home/ubuntu/libredwg/src'
Makefile:518: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/home/ubuntu/libredwg/src'
Makefile:672: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/ubuntu/libredwg'
Makefile:555: recipe for target 'all' failed
make: *** [all] Error 2
```

rurban, 2022-05-02:
gcc-7, which version exactly?

overide, 2022-05-03:
Hi, 

gcc version - gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0

libredwg tag - 0.12.4.4553

rurban, 2022-05-03:
Nice. I'll check with docker. gcc-7 is not the best, but gcc-9 is worse

overide, 2022-05-03:
Thanks, can you tell me what is the correct/safe way of using `libredwg` in a production environment? For my experimentation, I pulled the latest master branch but in prod, should I be using releases mentioned here? - https://github.com/LibreDWG/libredwg/releases/tag/0.12.5

rurban, 2022-05-03:
exactly

rurban, 2022-05-03:
I can only guess that your EC2 has restricted memory and/or stack settings. decode.o is really large. Worked fine in my bionic docker with gcc-7 and this tag.

rurban, 2022-05-03:
In the latest branch I split it up a bit, but it is not yet stable.

overide, 2022-05-04:
Thanks for the update, how much memory is required for `decode.o`? I was using `t2.micro` instance which has 1GB of memory, should I try it on higher memory instances?

closed 2022-12-30 by rurban