LibreDWG/libredwg #467: LibreDWG build on Ubuntu 20

AndreiCherniaev, 2022-05-09:
I try to build LibreDWG for my Ubuntu 20 using [instruction](https://wiki.freecadweb.org/FreeCAD_and_DWG_Import):
git clone https://git.savannah.gnu.org/git/libredwg.git
cd libredwg
mkdir build
cd build
cmake ..
make
make install (or use checkinstall, or simply locate & copy the dwg2dxf utility to your executables path, it will be then autodetected by FreeCAD)

But there is error during building:
_[ 25%] Building C object CMakeFiles/redwg.dir/src/in_json.c.o
/home/a/snap/freecad/common/libredwg/src/in_json.c:54:10: fatal error: ../jsmn/jsmn.h: No such file or directory
   54 | #include "../jsmn/jsmn.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated._

Looks like I forgot install some dependents on my system. But I want to say that it will be perfect if buildsystem checked which dependents are not is user's system now, istn't it?

rurban, 2022-05-10:
you missed ./autogen.sh, see the README
you can also check how the Docker or CI builds are doing it.

closed 2022-05-10 by rurban

rurban, 2022-05-10:
./configure --enable-trace --disable-write --disable-shared
does not work yet.

try ./configure or any configuration from the CI's. e.g. from `.github/workflows/main.yml`, `.cirrus.yml` or `.appveyor.yml`

AndreiCherniaev, 2022-05-10:
> ./configure --enable-trace --disable-write --disable-shared does not work yet.
> 
> try ./configure or any configuration from the CI's. e.g. from `.github/workflows/main.yml`, `.cirrus.yml` or `.appveyor.yml`

I use empty folder and try
sh ./autogen.sh (if you checked out the source code from git)
./configure
make -j3
And there is error.

But works after 
sudo apt-get install texinfo
I can build now.

AndreiCherniaev, 2022-05-10:
After build I try execute
dwg2dxf
_dwg2dxf: error while loading shared libraries: libredwg.so.0: cannot open shared object file: No such file or directory_

But I do 
sudo ldconfig

And now if I call dwg2dxf I get
_Usage: dwg2dxf [-v[N]] [--as rNNNN] [-m|--minimal] [-b|--binary] DWGFILES..._

But if I use FreeCAD I still got 
 _ODA (formerly Teigha) File Converter not found, DWG support is disabled_

[log.txt](https://github.com/LibreDWG/libredwg/files/8663935/log.txt)
@rurban

chinhuat, 2022-12-22:
> But if I use FreeCAD I still got _ODA (formerly Teigha) File Converter not found, DWG support is disabled_

That's not a libredwg issue, have you configured FreeCAD to use to libredwg? See: https://wiki.freecadweb.org/Import_Export_Preferences#DWG

sharyuke, 2024-10-25:
> After build I try execute dwg2dxf _dwg2dxf: error while loading shared libraries: libredwg.so.0: cannot open shared object file: No such file or directory_
>
> But I [...]

That's the solution for me.