LibreDWG/libredwg #468: Parsing of preR13 dimensions

michal-josef-spacek, 2022-05-09:
@rurban 
All dimensions in preR13 are one entity, which has special flag inside which differentiate dimension types.

Structure of dimension is:
```
flag_r11
entity_size
layer_index
opts_r11 (2 bytes, probably better is to use two flags)
... <------- elevation, thickness, handle, paper if flag_r11 flags
block_index
def_pt
text_midpt
flag1 <------ dimension type
// Other is defined by opts_r11 flags
```

michal-josef-spacek, 2022-05-09:
This https://github.com/michal-josef-spacek/libredwg/commit/06298c91f22cbbeb0de7ab67b87ec8228d9fdf71 detection is working, but i don't like it.

rurban, 2022-05-10:
Yes, I saw that logic from the kaitai files. However, we need to separate it beforehand into our types, and then fix the fields, and R11OPTS checks

michal-josef-spacek, 2022-05-10:
@rurban I understand.
But yours commit https://github.com/LibreDWG/libredwg/commit/5f1a43d9e7c6b8fe1f17df638278d0912781f92d is bad in case of dimension type switch on flag_r11.

rurban, 2022-05-10:
I only have examples for LINEAR and ORDINATE, but should be fixed in work/preR13

michal-josef-spacek, 2022-05-10:
@rurban  I am in progress of creating test file for dimensions. Wait a minute.

rurban, 2022-05-10:
Hmm, so with r2.6 the flag_r11 is not the type, but the opts_r11 flags. But DIA and RAD both have 0x22

michal-josef-spacek, 2022-05-10:
@rurban flag_r11 is not type at all.
flag_r11 in mine:
```
  entity_mode:
    seq:
      - id: entity_mode1
        type: b1
      - id: entity_pspace_flag
        type: b1
      - id: entity_handling_flag
        type: b1
      - id: entity_mode4
        type: b1
      - id: entity_thickness_flag
        type: b1
      - id: entity_elevation_flag
        type: b1
      - id: entity_linetype_flag
        type: b1
      - id: entity_color_flag
        type: b1
```
This is RC

After this are:
entity_size RS
entity_layer_index RS
opts_flag1 RC
opts_flag2 RC

After this are:
if _some_flags_r11_
  some parsing
* color
* linetype
* elevation
* thickness
* pspace
* vport
* handling
* probably other defined in flags_r11 flag

------ End of common entity block ------

After this in dimensions:
block_index RS
def_pt 2D|3D
text_midpt 2D|3D

dim_type RC

Everything releated to dim_type flags.

michal-josef-spacek, 2022-05-10:
sorry dim_type is more complex:
```
  dim_type:
    seq:
      - id: flag_text_in_user_location
        type: b1
      - id: flag_x_type_ordinate
        type: b1
      - id: flag_block_for_dim_only
        type: b1
      - id: flag_u4
        type: b1
      - id: type
        type: b4
        enum: dim_type
```

rurban, 2022-12-26:
It's better now, but still not 100% correct

rurban, 2022-12-30:
Now it's correct

closed 2022-12-30 by rurban