LibreDWG/libredwg #474: dxf2dwg file does not open

wanfahmi11, 2022-05-14:
Hello everyone,

I’m using the dxf2dwg program and converting does prints some errors and warning:

ERROR: dwg_dynapi_entity_set_value: Invalid VPORT_CONTROL field objid
ERROR: dwg_dynapi_entity_set_value: Invalid LTYPE_CONTROL field objid
ERROR: dwg_dynapi_entity_set_value: Invalid LAYER_CONTROL field objid
ERROR: dwg_dynapi_entity_set_value: Invalid STYLE_CONTROL field objid
ERROR: dwg_dynapi_entity_set_value: Invalid VIEW_CONTROL field objid
ERROR: dwg_dynapi_entity_set_value: Invalid UCS_CONTROL field objid
ERROR: dwg_dynapi_entity_set_value: Invalid APPID_CONTROL field objid
ERROR: dwg_dynapi_entity_set_value: Invalid DIMSTYLE_CONTROL field objid
ERROR: dwg_dynapi_entity_set_value: Invalid BLOCK_CONTROL field objid
Warning: Unknown HEADER.DIMSTYLE STANDARD dxf:3
Warning: Unknown HEADER.CMLSTYLE STANDARD dxf:2
Warning: Unknown HEADER.DIMSTYLE STANDARD dxf:3
Warning: Object handle not found, 1/1 in 41156 objects
Warning: Object handle not found, 12/C in 41156 objects

The program successfully outputs to a DWG file. However, I got “Drawing file was created by an incompatible version” when opening it on Autodesk DWG Trueview 2022. Any clue? Thanks in advance

michal-josef-spacek, 2022-05-28:
@wanfahmi11 Could be good insert example dxf file (something.dxf.gz/.zip).

wanfahmi11, 2022-09-06:
Here's the DXF file. Sorry for the delay. I missed this out. Thank you in advance and thank you for this amazing library!

[my-drawing.dxf.zip](https://github.com/LibreDWG/libredwg/files/9494937/my-drawing.dxf.zip)

michal-josef-spacek, 2022-09-06:
@wanfahmi11 Thanks. 
BTW: There is no DXF version in DXF file. What is generating this kind of DXF files?

wanfahmi11, 2022-09-06:
The DXF is generated through the Maker.js library https://maker.js.org/docs/exporting/#content. The DXF is very minimal though but still can be opened in Autodesk DWG TrueView.

michal-josef-spacek, 2022-09-06:
@wanfahmi11 Thank you for the information.

rurban, 2022-09-09:
The problem is that such minimal DXF contain no handles, and we use the handles internally to find table entries. Thus we need to assign handles ourselves, at least on table controls and its entries.