LibreDWG/libredwg #483: preR13 block table entry

michal-josef-spacek, 2022-06-02:
@rurban
In the BLOCK_HEADER table entry is after name address of entity in block list.

How could I process this address?

michal-josef-spacek, 2023-01-03:
[BLOCK.DWG.gz](https://github.com/LibreDWG/libredwg/files/10337809/BLOCK.DWG.gz)

In some situations we have situation:
```
blocks_start: 4251 (0x109b) [RL]
blocks_size: 2264 [RL]
blocks_end: 0 (0x0) [RL]
blocks_max: 0x80000000 [RLx]
==========================================
ptr table BLOCK    [ 1]: size:38   num:2  (0xb29-0xb75) flags:0x8011
```
Where blocks_end isn't defined and there is the only size of a block.

The output of block entities is bad like this:
```
block entities: (0x109b-0x0 (0), size 2264, 0x80000000)
==========================================
```

michal-josef-spacek, 2023-01-05:
Example files:
[BLOCK.scr.gz](https://github.com/LibreDWG/libredwg/files/10348621/BLOCK.scr.gz)
[BLOCK.DXF.gz](https://github.com/LibreDWG/libredwg/files/10348623/BLOCK.DXF.gz)
[BLOCK.DWG.gz](https://github.com/LibreDWG/libredwg/files/10348624/BLOCK.DWG.gz)

There are two blocks.
In block table is information about start address of block entity in block entities section.
Example:
Two blocks:
```
ptr table BLOCK    [ 1]: size:37   num:2  (0x2e5-0x32f) flags:0x8010
...
-- table entry BLOCK_HEADER [0]: 0x2e5
flag: 0 [RC 70]
name: "ABCDEFGHIJKLMNOPQRSTUVWXYZ12345" [TF 32 2]
block_scaling: 0x0 [RC 0]
num_owned: 0 [RS 0]
flag2: 0x0 [RC 0]
  ADD_OBJECT BLOCK_HEADER [9]
  handle 0.1.23
BLOCK_HEADER.name = BLOCK
BLOCK_CONTROL.entries[1] = (2.1.23) abs:23

-- table entry BLOCK_HEADER [1]: 0x30a
flag: 0 [RC 70]
name: "BLOCK" [TF 32 2]
block_scaling: 0x48 [RC 0]
num_owned: 0 [RS 0]
flag2: 0x0 [RC 0]
contents table LAYER    [ 2]: size:37   num:1   (0x32f-0x354)
  ADD_OBJECT LAYER [10]
  handle 0.1.24
LAYER.name = 0
LAYER_CONTROL.entries[0] = (2.1.24) abs:24
```

Field block_scaling is address to block entity in block entities section.

Question in this issue was unknown, but we need cleanup this.

rurban, 2023-01-05:
yes, block_scaling looks wrong. needs to be renamed to block_index or block_address or block_offset or such

michal-josef-spacek, 2023-01-10:
I looked at it, I don't know still.
I don't understand what are ```num_owned```, ```used``` and ```flag2```. 
Do you have some additional information or examples or know-how related to this?

michal-josef-spacek, 2023-01-11:
I have some result. It's simple.

Only issue are 4 bytes which are ffffffff on R11 and something like example in R14 output to R11

Example 4 bytes:
```
58 05 4f 07
```
Could be a handle on preR13?

rurban, 2023-01-11:
used -1 is pretty common. r11 handles are usually only 1 or 2 bytes I thought

michal-josef-spacek, 2023-01-11:
https://github.com/LibreDWG/libredwg/pull/568

michal-josef-spacek, 2023-01-11:
TODO: encoding
TODO: identify R_11 fields

I am for merge still.

rurban, 2023-01-20:
Other than those 2 TODO's this is now done, right? We can even decode block entities now.

michal-josef-spacek, 2023-01-20:
@rurban Yes, we are decoding.

rurban, 2023-01-20:
I'm refactoring this now a bit.
block_addr => block_offset_r11,
and moving the entities|blocks_ start/end/siye, ... fields to the dwg->header

rurban, 2023-01-20:
Fixed

closed 2023-01-20 by rurban