LibreDWG/libredwg #485: A NULL pointer dereference exists in the function decode_preR13 in decode_r11.c

cxlzff, 2022-06-07:
### system info
Ubuntu x86_64, clang 6.0, dwg2dxf([0.12.4.4608](https://github.com/LibreDWG/libredwg/releases/tag/0.12.4.4608))

### Command line
./programs/dwg2dxf -b -m @@ -o /dev/null

### AddressSanitizer output
==8985==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x7ffff6f8bc74 bp 0x7fffffffca90 sp 0x7fffffffc1f8 T0)
==8985==The signal is caused by a READ memory access.
==8985==Hint: address points to the zero page.
    #0 0x7ffff6f8bc73  /build/glibc-CVJwZb/glibc-2.27/string/../sysdeps/x86_64/multiarch/memcmp-avx2-movbe.S:267
    #1 0x49db15 in __interceptor_memcmp.part.75 /fuzzer/build/llvm_tools/llvm-4.0.0.src/projects/compiler-rt/lib/asan/../sanitizer_common/sanitizer_common_interceptors.inc:711
    #2 0x705a4d in decode_preR13 /testcase/libredwg/src/decode_r11.c:771:11
    #3 0x53245a in dwg_decode /testcase/libredwg/src/decode.c:209:23
    #4 0x50d759 in dwg_read_file /testcase/libredwg/src/dwg.c:254:11
    #5 0x50c454 in main /testcase/libredwg/programs/dwg2dxf.c:258:15
    #6 0x7ffff6e22c86 in __libc_start_main /build/glibc-CVJwZb/glibc-2.27/csu/../csu/libc-start.c:310
    #7 0x419ee9 in _start (/testcase/libredwg/programs/dwg2dxf+0x419ee9)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /build/glibc-CVJwZb/glibc-2.27/string/../sysdeps/x86_64/multiarch/memcmp-avx2-movbe.S:267 
==8985==ABORTING

### poc
https://gitee.com/cxlzff/fuzz-poc/raw/master/libredwg/decode_preR13_np

rurban, 2022-06-24:
See GH #486

rurban, 2022-12-15:
Fails now with ERROR: Wrong HEADER Section Locator Records at 20

closed 2022-12-15 by rurban