LibreDWG/libredwg #486: A NULL pointer dereference exists in the function decode_preR13_section in decode_r11.c

cxlzff, 2022-06-07:
### system info
Ubuntu x86_64, clang 6.0, dwg2dxf([0.12.4.4608](https://github.com/LibreDWG/libredwg/releases/tag/0.12.4.4608))

### Command line
./programs/dwg2dxf -b -m @@ -o /dev/null

### AddressSanitizer output
==8987==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000048 (pc 0x00000070e7b8 bp 0x7fffffffca90 sp 0x7fffffffc8c0 T0)
==8987==The signal is caused by a READ memory access.
==8987==Hint: address points to the zero page.
    #0 0x70e7b7 in decode_preR13_section /testcase/libredwg/src/decode_r11.c:399:15
    #1 0x705d5e in decode_preR13 /testcase/libredwg/src/decode_r11.c:833:12
    #2 0x53245a in dwg_decode /testcase/libredwg/src/decode.c:209:23
    #3 0x50d759 in dwg_read_file /testcase/libredwg/src/dwg.c:254:11
    #4 0x50c454 in main /testcase/libredwg/programs/dwg2dxf.c:258:15
    #5 0x7ffff6e22c86 in __libc_start_main /build/glibc-CVJwZb/glibc-2.27/csu/../csu/libc-start.c:310
    #6 0x419ee9 in _start (/testcase/libredwg/programs/dwg2dxf+0x419ee9)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /testcase/libredwg/src/decode_r11.c:399:15 in decode_preR13_section
==8987==ABORTING

### poc
https://gitee.com/cxlzff/fuzz-poc/raw/master/libredwg/decode_preR13_section_np

rurban, 2022-06-24:
BTW: Decoding preR13 DWG's was never released.
0.12.5:
```
Reading DWG file ../test/issues/gh486/decode_preR13_section_np
ERROR: This version of LibreDWG is only capable of decoding version r13-r2018 (code: AC1012-AC1032) DWG files.
We don't decode many entities and no blocks yet.
ERROR: Unknown object type 0
ERROR: Invalid table number 65548 for STYLE    [ 3]
ERROR: Invalid table number 128 for LTYPE    [ 5]
ERROR: Invalid table number -16777216 for VIEW     [ 6]
ERROR: Invalid table number 117901063 for UCS      [ 7]
ERROR: Invalid table number -33159417 for VPORT    [ 8]
ERROR: Invalid table number 117901063 for APPID    [ 9]
ERROR: Invalid table number 117901063 for DIMSTYLE [10]
ERROR: Invalid table number 67372036 for VX       [11]
ERROR: Failed to decode file: ../test/issues/gh486/decode_preR13_section_np 0x800

READ ERROR 0x800
```

rurban, 2022-12-15:
Fixed with "preR13: better wrong dat->chain free fix"