LibreDWG/libredwg #491: bad-free exists in the function dwg_read_file in dwg.c

cxlzff, 2022-06-07:
### system info
Ubuntu x86_64, clang 6.0, dwg2dxf([0.12.4.4608](https://github.com/LibreDWG/libredwg/releases/tag/0.12.4.4608))

### Command line
./programs/dwg2dxf -b -m @@ -o /dev/null

### AddressSanitizer output
==8999==ERROR: AddressSanitizer: attempting free on address which was not malloc()-ed: 0x61a0000000e0 in thread T0
    #0 0x4d23a0 in __interceptor_cfree.localalias.0 /fuzzer/build/llvm_tools/llvm-4.0.0.src/projects/compiler-rt/lib/asan/asan_malloc_linux.cc:55
    #1 0x50d77a in dwg_read_file /testcase/libredwg/src/dwg.c:258:7
    #2 0x50c454 in main /testcase/libredwg/programs/dwg2dxf.c:258:15
    #3 0x7ffff6e22c86 in __libc_start_main /build/glibc-CVJwZb/glibc-2.27/csu/../csu/libc-start.c:310
    #4 0x419ee9 in _start (/testcase/libredwg/programs/dwg2dxf+0x419ee9)

0x61a0000000e0 is located 96 bytes inside of 1309-byte region [0x61a000000080,0x61a00000059d)
allocated by thread T0 here:
    #0 0x4d2750 in calloc /fuzzer/build/llvm_tools/llvm-4.0.0.src/projects/compiler-rt/lib/asan/asan_malloc_linux.cc:74
    #1 0x50cdd0 in dat_read_file /testcase/libredwg/src/dwg.c:91:33
    #2 0x50d708 in dwg_read_file /testcase/libredwg/src/dwg.c:247:15
    #3 0x50c454 in main /testcase/libredwg/programs/dwg2dxf.c:258:15
    #4 0x7ffff6e22c86 in __libc_start_main /build/glibc-CVJwZb/glibc-2.27/csu/../csu/libc-start.c:310

SUMMARY: AddressSanitizer: bad-free /fuzzer/build/llvm_tools/llvm-4.0.0.src/projects/compiler-rt/lib/asan/asan_malloc_linux.cc:55 in __interceptor_cfree.localalias.0
==8999==ABORTING

### poc
https://gitee.com/cxlzff/fuzz-poc/raw/master/libredwg/dwg_read_file_baffree

rurban, 2022-12-15:
This is caused by a changed dat->chain ptr in decode_r11.c:396 (bit_reset_chain) and a error return, skipping the chain restore.
We should really use the dwg.spec _private function for that

rurban, 2022-12-15:
Fixed this case by changing FIELD_FD to FIELD_VECTOR_INL, but ultimately we need to use dwg.spec for preR13 tables.

closed 2022-12-15 by rurban