LibreDWG/libredwg #498: Cannot build libredwg with --disable-write

michal-josef-spacek, 2022-06-24:
I tested on smoke/preR13 and work/preR13 branches

```
$ make distclean
$ ./configure --disable-write
$ make
```

Error:
```
...
  CCLD     dwgread
/usr/bin/ld: ../src/.libs/libredwg.so: undefined reference to `dwg_add_LAYER'
/usr/bin/ld: ../src/.libs/libredwg.so: undefined reference to `dwg_add_STYLE'
/usr/bin/ld: ../src/.libs/libredwg.so: undefined reference to `dwg_add_APPID'
/usr/bin/ld: ../src/.libs/libredwg.so: undefined reference to `dwg_add_LTYPE'
/usr/bin/ld: ../src/.libs/libredwg.so: undefined reference to `dwg_add_VIEW'
/usr/bin/ld: ../src/.libs/libredwg.so: undefined reference to `dwg_add_DIMSTYLE'
/usr/bin/ld: ../src/.libs/libredwg.so: undefined reference to `dwg_add_VPORT'
/usr/bin/ld: ../src/.libs/libredwg.so: undefined reference to `dwg_add_BLOCK_HEADER'
/usr/bin/ld: ../src/.libs/libredwg.so: undefined reference to `dwg_add_Document'
collect2: error: ld returned 1 exit status
...
```

rurban, 2022-06-24:
yeah, this might deserve a github actions entry

michal-josef-spacek, 2022-06-25:
https://github.com/LibreDWG/libredwg/pull/499

rurban, 2022-07-31:
Fixed

closed 2022-07-31 by rurban

michal-josef-spacek, 2022-07-31:
Great, it's working. Thank you.