LibreDWG/libredwg #513: Wrong layer indexed colors for dxf2004,2007,2010,2013,2018

zamtmn, 2022-07-31:
layer indexed colors works only in  dxf2000
[dxf_colors.zip](https://github.com/LibreDWG/libredwg/files/9229233/dxf_colors.zip)

rurban, 2022-11-23:
Can you please describe what doesn't work? the layers do have the right colors.
Is dwg2dxf resp. dxf2dxf producing wrong results with LAYER.color fields?

timoria21, 2023-03-09:
Hi,

With a file of mine, I had a similar issue: for a given layer I was expecting ACI index 153, instead I had 174.

I noticed that by just taking RGB & 0x00ff0000 in the `bit_read_CMC()` method instead of calling `dwg_find_color_index(color->rgb)` I get the right result; this solved some issues with another file of mine as well.

Is this by chance or is there a reason that justifies it?

Thanks,

Dave

rurban, 2023-06-27:
Should be fixed with commit 21a9f24ab9e914823aa1c6da7bb0980520d076c9 (tag: 0.12.5.5852)
Author: Reini Urban <rurban@cpan.org>
Date:   Sat Jun 24 10:39:02 2023 +0200

    encode: fix bit_downconvert_CMC
    
    for non-truecolor color index

closed 2023-06-27 by rurban