LibreDWG/libredwg #519: Assertion failed in binary dwglayers

Yhcrown, 2022-08-20:
Hi, developers of libredwg:
I also find an assertion failed in binay **dwglayers**. The input and the information are as follows. Any help would be greatly appreciated from you :D
**input:**
https://github.com/Yhcrown/fuzzing-crashes/blob/main/validated/dwglayers/0
**output:**
```
dwglayers: dwglayers.c:166: int main(int, char **): Assertion `_ctrl->entries[i]' failed.
```
**enviroment:**
Ubuntu 22.04 LTS
Linux lab-pro 4.15.0-147-generic
gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0
Ubuntu clang version 14.0.6

**Thanks & Best regards !**

rurban, 2022-08-21:
Invalid input, caused by fuzzing obviously.
The link isr3sd, so I cannot check it

Yhcrown, 2022-08-21:
> Invalid input, caused by fuzzing obviously.
> The link is dead, so I cannot check it

Sorry, I've updated.

michal-josef-spacek, 2022-09-10:
FYI: Autocad R14 cannot open this file.

closed 2022-12-01 by rurban