LibreDWG/libredwg #520: issue in reading dwg and convert arc drawings to geojson

mahmoudadel54, 2022-10-12:
I have a dwg with drawing arcs. I tried to use dwgread.exe with GeoJSON output, it gives me null in geometry

[image]

Please If someone can help !!

rurban, 2022-10-14:
See out_geojson.c:775
```
      // needs explosion of arc into lines
      // dwg_geojson_ARC(dat, obj);
      LOG_TRACE ("ARC not yet supported")
```

mahmoudadel54, 2022-10-19:
> See out_geojson.c:775
> 
> ```
>       // needs explosion of arc into lines
>       // dwg_geojson_ARC(dat, obj);
>       LOG_TRACE ("ARC not yet supported")
> ```

Actually I have no experience with C language

rurban, 2022-10-19:
I started working on it in the work/arcsplit branch

closed 2023-08-13 by rurban