LibreDWG/libredwg #521: Failed to install on fresh Ubuntu installation with python3.8: Could not link test program to Python

BastienCagna, 2022-10-17:
Hello everybody,
I just re-install ubuntu 18 on my lap top and try to install libredwg. I done apt update and upgrade after installation, then installed python3.8 and  pip and set it as default by editing the simlink and installed following packages: git autoconf libtool swig python-dev texinfo build-essential gcc
Then, I have cloned the git repo, run successfully autogen.sh but when it comes to ./configure, I got some errors concerning python:

checking python extra linking flags... -Xlinker -export-dynamic -Wl,-O1 -Wl,-Bsymbolic-functions
checking consistency of all components of python development environment... no
configure: error: in `/home/bastien/soft/libredwg':
configure: error: 
  Could not link test program to Python. Maybe the main Python library has been
  installed in some non-standard library path. If so, pass it to configure,
  via the LIBS environment variable.
  Example: ./configure LIBS="-L/usr/non-standard-path/python/lib"
  ============================================================================
   ERROR!
   You probably have to install the development version of the Python package
   for your distribution.  The exact name of this package varies among them.
  ============================================================================

See `config.log' for more details

I'm a bit confused as I didn't do anything special since the installation of ubuntu (I'd just install python 3.8 and replace /usr/bin/python to a symlink to python3.8). 
Does some one have an idea about what's going wrong there?

rurban, 2022-10-18:
No idea, the python-dev probe is constantly changing.Try with --disable-bindings

BastienCagna, 2022-10-18:
Thanks @rurban for your answer. I tried and the compilation seems to work but my goal is precisely to use the python bindings...
I didn't really understand what is the problem about python.

BastienCagna, 2022-10-18:
Installing python3.8-dev resolved the problem! =D

```shell
sudo apt-get install python3.8-dev
```

closed 2022-10-18 by BastienCagna