LibreDWG/libredwg #522: How should I access to object's layer name?

BastienCagna, 2022-10-19:
Hello all,
## Context 
I recently forked the project and started to work in examples/dwg2svg.C to cover some my needs. I want to export architectural plans and keep the layers organization (either groups in svg or a layer-name attribute in each tags). 
I reached to add lwpolyline and ellipse export but I'm facing issues to get the layer name.

## Issue

```c
Dwg_Object_LAYER *layer = dwg_get_entity_layer(poly->parent);
fprintf (stderr, "%s\n", layer->name);
```
This print only the first letter of the name when the next two options go in segmentation fault:
```c
  //char *utf8 = bit_convert_TU (dwg_ent_get_layer_name(poly, &error));
  //char *name =dwg_ent_get_layer_name(obj->parent, &error);
```

As I didn't use C for many years, I think that I'm doing something wrong but don't no what. :'(

rurban, 2022-10-20:
Since r2007 DWG's all texts will be unicode.
use dwg_ent_get_layer_name() instead, which converts it automatically to UTF-8.

BastienCagna, 2022-10-20:
Ok thanks @rurban!

Looking to testsuite.c, I found the solution with dwg_ent_get_layer_name().
Here it is:
```c
char *name = dwg_ent_get_layer_name(obj->tio.entity, &error);
```

closed 2022-10-20 by BastienCagna