LibreDWG/libredwg #532: Entity type id

michal-josef-spacek, 2022-12-09:
Hi @rurban 
When we are decoding e.g. AC1.4 in Entity obj->type will be a number of entity in R11 (LINE = 1).
When we are decoding e.g. AC1015 in Entity obj->type will be a number of entities in > R11 (LINE = 19).

What about the unification of this type to the > R11 version?
I think that we could do better conversions between versions after the change.

Opposite example:
When I use dwg_add_LINE, the type id will be 19 in every version. And I couldn't write this type to AC1.4 DWG file, because is bad.

What do you think?
I do some changes but I don't know what we want.

rurban, 2022-12-11:
No, the recent acad version should be the default.
Thats why we convert the dwg type to a global fiyed_type, with all the dynamic classes

michal-josef-spacek, 2022-12-11:
Ok, then there is PR (https://github.com/LibreDWG/libredwg/pull/534) for a fix of encoding for R11

rurban, 2022-12-12:
Merged

closed 2022-12-12 by rurban