LibreDWG/libredwg #535: dwgadd issue in text entity

michal-josef-spacek, 2022-12-12:
In input:
```
text "test" (0.0 1.0 0.0) 8
```
dwgadd badly parse text string to `test"`

rurban, 2022-12-12:
Yes, I just fixed this %119s artefact. It parses until the next whitespace

rurban, 2022-12-12:
Fixed with 8936c5e44e46b03008b4a5e7944850e60784fae4

michal-josef-spacek, 2022-12-12:
After these changes, pre13 support not working again :-)
None of the entities was written.

michal-josef-spacek, 2022-12-12:
Debug with dwg_add_LINE to AC1.4:
```
Skip entity LINE, number: 9, Fixedtype: 19, Addr: 0 (0x202)
```

rurban, 2022-12-13:
Should work now

michal-josef-spacek, 2022-12-13:
> Should work now

No commit, i couldn't test :-)

rurban, 2022-12-13:
This LINE is a deleted LINE which was moved into a block.

It is encoded as such:
```
type: 65535 [RS]
Add entity LINE [17] Decode entity LINE
===========================
Entity number: 17, Type: 255, Addr: 320
layer: 1 [H(RSd) 8]
start: (6.000000, 1.000000) [2RD 10]
end: (7.000000, 2.000000) [2RD 11]
```

michal-josef-spacek, 2022-12-13:
Hm, I missed something.

When i dwgadd --as r1.4 this:
```
text "test" (0.0 1.0 0.0) 8
```
Output is ok. Instead of issues with values in it (no right point and height)

When I dwgadd --as r1.4 this:
```
line (0 0 0) (100 100 100)
```
There is no entity.
Output of dwgadd:
```
...
SKETCHINC: 0.100000 [RD 40] @506.0
FILLETRAD: 0.000000 [RD 40] @514.0

entities 0x202:
Skip object BLOCK_CONTROL, number: 0, Fixedtype: 48, Addr: 0 (0x202)
Skip object LAYER_CONTROL, number: 1, Fixedtype: 50, Addr: 0 (0x202)
Skip object STYLE_CONTROL, number: 2, Fixedtype: 52, Addr: 0 (0x202)
Skip object LTYPE_CONTROL, number: 3, Fixedtype: 56, Addr: 0 (0x202)
Skip object VIEW_CONTROL, number: 4, Fixedtype: 60, Addr: 0 (0x202)
Skip object BLOCK_HEADER, number: 5, Fixedtype: 49, Addr: 0 (0x202)
Skip entity BLOCK, number: 6, Fixedtype: 4, Addr: 0 (0x202)
Skip entity ENDBLK, number: 7, Fixedtype: 5, Addr: 0 (0x202)
Skip entity LINE, number: 8, Fixedtype: 19, Addr: 0 (0x202)

entities 0 0x202 - 0x202
0x24: num_bytes: 514 [RL]
Wrote 514 bytes

============
dwg_free
Free object BLOCK_HEADER [5]
Free entity BLOCK [6]
Free entity ENDBLK [7]
Free entity LINE [8]
Free object BLOCK_CONTROL [0]
Free object LAYER_CONTROL [1]
Free object STYLE_CONTROL [2]
Free object LTYPE_CONTROL [3]
Free object VIEW_CONTROL [4]
free 13 global refs
```

Inside of dwg file is:
```
...
dwg_size: 514 [RL 0]     # Only header
numentities: 0 [RS 0]
...
entities: (0x202-0x202 (0), size 0, 0x0)
==========================================

============
```

michal-josef-spacek, 2022-12-13:
Everything on master branch

rurban, 2022-12-13:
See branch `smoke/add-r1.4`
Several critical header vars were still missing.

rurban, 2022-12-13:
Text is fixed, the rest is coming along

closed 2022-12-13 by rurban