LibreDWG/libredwg #540: Issue with reading of numheader_vars

michal-josef-spacek, 2022-12-18:
Commit d6029e974 with changes in dwg_add_Document() is bad.

Decoding process is:
* read numheader_vars from DWG file
* dwg_add_Document (which after this commit set this variable to different values)
* usage of numheader_vars

As I understand, we need to move this setting to dwg_new_Document()

rurban, 2022-12-18:
Looks like so, indeed.. Maybe just set it there if it's empty

michal-josef-spacek, 2022-12-18:
I believe that isn't empty.

michal-josef-spacek, 2022-12-18:
Example which is not working
[3DFACE1.DWG.gz](https://github.com/LibreDWG/libredwg/files/10253686/3DFACE1.DWG.gz)

rurban, 2022-12-19:
I see the problem now. decode_r11 needs to call add_Document because of all the tables, but we must not ignore the existing numheader_vars. Unless we change the target version.

michal-josef-spacek, 2022-12-22:
It's working, thanks.

closed 2022-12-22 by michal-josef-spacek