LibreDWG/libredwg #542: Fix polyline flag in preR13

michal-josef-spacek, 2023-01-01:
In https://github.com/LibreDWG/libredwg/blob/master/src/dwg.spec#L1167 we have:
```
    if (R11OPTS (1))
      FIELD_RC (has_vertex, 66);
```
This is bad, it's
```
    if (R11OPTS (1))
      FIELD_RC (flag, 70);
```
Same flag as https://github.com/LibreDWG/libredwg/blob/master/include/dwg.h#L1545:
```
BITCODE_BS flag;        /* 1: closed, 2: curve_fit, 4: spline_fit, 8: 3d, 0x10: 3dmesh,
                             0x20: mesh_closed_in_n, 0x40: polyface_mesh, 0x80: ltype_continuous */
```
Not short, but byte in preR13.
In preR13 are used only 1 (closed) and 2 (curve_fit) probably

rurban, 2023-01-02:
remaining question is R11OPTS(8) or 16 for curve_type (0-5)

michal-josef-spacek, 2023-01-02:
Thanks a lot.

> remaining question is R11OPTS(8) or 16 for curve_type (0-5)

I am looking at it systematically

closed 2023-01-02 by michal-josef-spacek