LibreDWG/libredwg #548: Issue with Polyline in AC1003

michal-josef-spacek, 2023-01-03:
[fit_curve_pline.scr.gz](https://github.com/LibreDWG/libredwg/files/10336468/fit_curve_pline.scr.gz)
[fit_curve_pline.dxf.gz](https://github.com/LibreDWG/libredwg/files/10336469/fit_curve_pline.dxf.gz)
[fit_curve_pline.dwg.gz](https://github.com/LibreDWG/libredwg/files/10336470/fit_curve_pline.dwg.gz)

Something wierd:
```
entities: (0x3d5-0x431 (7), size 92, 0x0)
==========================================
type: 18 [RCd]
Add entity POLYLINE_2D [8] Decode entity POLYLINE_2D
===========================
Entity number: 8, Type: 18, Addr: 3d5
flag_r11: 0x0 [RC 0]
size: 8 [RS]
layer: 146 [H(RSd) 8]
opts_r11: 0x8000 [RSx 0]
```
1) layer: 146, sic.
2) opts_r11: 0x8000

Second polyline:
```
type: 18 [RCd]
Add entity POLYLINE_2D [9] Decode entity POLYLINE_2D
===========================
Entity number: 9, Type: 18, Addr: 3dd
flag_r11: 0x0 [RC 0]
size: 24 [RS]
layer: 73 [H(RSd) 8]
opts_r11: 0x8000 [RSx 0]

ERROR: offset 16
```
Different size, same type.

michal-josef-spacek, 2023-01-03:
R11 output:

[fit_curve_pline_r11.dxf.gz](https://github.com/LibreDWG/libredwg/files/10337426/fit_curve_pline_r11.dxf.gz)
[fit_curve_pline_r11.dwg.gz](https://github.com/LibreDWG/libredwg/files/10337427/fit_curve_pline_r11.dwg.gz)

Same

michal-josef-spacek, 2023-01-03:
R14 output exported as R11:

[fit_curve_pline_r14_exported_as_r11.dwg.gz](https://github.com/LibreDWG/libredwg/files/10337461/fit_curve_pline_r14_exported_as_r11.dwg.gz)

Different:
```
entities: (0x6cf-0x79a (0), size 203, 0x0)
==========================================
type: 19 [RCd]
Add entity POLYLINE_2D [13] Decode entity POLYLINE_2D
===========================
Entity number: 13, Type: 19, Addr: 6cf
flag_r11: 0xa0 [RC 0]
          HAS_HANDLING (0x32) HAS_ATTRIBS (0x128)
size: 13 [RS]
layer: 0 [H(RSd) 8]
opts_r11: 0x1 [RSx 0]
          HAS_POLYLINE_FLAG (1)
handle: 0.1.20 [H 5]
flag: 0x2 [RC 70]
      CURVE_FIT (2)
crc: 540C [RSx]

type: 20 [RCd]
Add entity VERTEX_2D [14] Decode entity VERTEX_2D
===========================
Entity number: 14, Type: 20, Addr: 6dc
flag_r11: 0x20 [RC 0]
          HAS_HANDLING (0x32)
size: 36 [RS]
layer: 0 [H(RSd) 8]
opts_r11: 0x4 [RSx 0]
          HAS_BULGE (4)
handle: 0.1.21 [H 5]
point: (1.000000, 1.000000) [2RD 10]
bulge: -0.198912 [RD 42]
crc: 7429 [RSx]
```

rurban, 2023-01-04:
> r11
did you run the script in r11 as such, or did you just load the broken r2.6 dwg there? I assume the latter, which would explain the still broken dwg there.
good would be a new r11 dwg with the curve fit.

michal-josef-spacek, 2023-01-04:
about first r11 version)
I generated dwg file from scr file in R11.

michal-josef-spacek, 2023-01-04:
r10 version:

[fit_curve_pline.dxf.gz](https://github.com/LibreDWG/libredwg/files/10342184/fit_curve_pline.dxf.gz)
[fit_curve_pline.dwg.gz](https://github.com/LibreDWG/libredwg/files/10342185/fit_curve_pline.dwg.gz)

rurban, 2023-01-04:
I check now for opts_r11 0x8000 and change it to a EXTRA_VERTEX 2D (when the size matches)

rurban, 2023-01-04:
Done

michal-josef-spacek, 2023-02-09:
This is not a right solution.
0x8000 flag mean, that the entity is replaced by entity in extra entities section.
PR to fix: https://github.com/LibreDWG/libredwg/pull/634

Output for fit_curve_pline.dwg above:
```
type: 18 [RCd]
Add entity POLYLINE_2D [8] Decode entity POLYLINE_2D
===========================
Entity number: 8, Type: 18, Addr: 3d5
flag_r11: 0x0 [RC 0]
size: 8 [RS]
layer: 146 [H(RSd) 8]
opts_r11: 0x8000 [RSx 0]
          REPLACED_IN_EXTRA(0x8000)

type: 18 [RCd]
Add entity POLYLINE_2D [9] Decode entity POLYLINE_2D
===========================
Entity number: 9, Type: 18, Addr: 3dd
flag_r11: 0x0 [RC 0]
size: 24 [RS]
layer: 73 [H(RSd) 8]
opts_r11: 0x8000 [RSx 0]
          REPLACED_IN_EXTRA(0x8000)
unknown (16): 44E087071A1905804900008018001180

type: 18 [RCd]
Add entity POLYLINE_2D [10] Decode entity POLYLINE_2D
===========================
Entity number: 10, Type: 18, Addr: 3f5
flag_r11: 0x0 [RC 0]
size: 24 [RS]
layer: 0 [H(RSd) 8]
opts_r11: 0x8000 [RSx 0]
          REPLACED_IN_EXTRA(0x8000)
unknown (16): 44E087071A1905800000008018001180
```

Output for fit_curve_pline_r11.dwg above:
```
type: 18 [RCd]
Add entity POLYLINE_2D [13] Decode entity POLYLINE_2D
===========================
Entity number: 13, Type: 18, Addr: 6cf
flag_r11: 0x0 [RC 0]
size: 10 [RS]
layer: 158 [H(RSd) 8]
opts_r11: 0x8000 [RSx 0]
          REPLACED_IN_EXTRA(0x8000)
crc: A7A3 [RSx]

type: 18 [RCd]
Add entity POLYLINE_2D [14] Decode entity POLYLINE_2D
===========================
Entity number: 14, Type: 18, Addr: 6d9
flag_r11: 0x0 [RC 0]
size: 26 [RS]
layer: 79 [H(RSd) 8]
opts_r11: 0x8000 [RSx 0]
          REPLACED_IN_EXTRA(0x8000)
unknown (16): 980B00000000F03F000000000000F03F
crc: E9C4 [RSx]

type: 18 [RCd]
Add entity POLYLINE_2D [15] Decode entity POLYLINE_2D
===========================
Entity number: 15, Type: 18, Addr: 6f3
flag_r11: 0x0 [RC 0]
size: 26 [RS]
layer: 0 [H(RSd) 8]
opts_r11: 0x8000 [RSx 0]
          REPLACED_IN_EXTRA(0x8000)
unknown (16): 8EDF0000000000400000000000000040
crc: F8E0 [RSx]
```

leshasoft, 2023-02-10:
Object 18 (0x12)  it looks like a pointer, if it is in the Entities section, then two bytes after the length offset in the section Extra Entities.
If it is in the section Extra Entities, then the offsets are not real, something else is possible.

```
           0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
    000500 F9 00 02 00 00 00 40 07-00 00 00 00 06 00 00 00   ......@.........
    000510 FF FF                                             ..              

ENTITIES
    000512 12 00    unknown, entprop 0H (0)
    000514 08 00    length 8H (8)
    000516 00 00 00 80    

    00051A 

    00051A 14 00    vertex, entprop 0H (0)
    00051C 18 00    length 18H (24)
    00051E 00 00    layer index 0H (0)
    000520 00 00    entflag 0H (0)
    000522 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00   ................
    000532 

....

EXTRA ENTS
    000932 13 80 09 00 00 00 01 00-01 12 00 08 00 1A 05 00   ................
    000942 00          
```                                      .

michal-josef-spacek, 2023-02-12:
@rurban This is not ok

Example:
[PL2.scr.gz](https://github.com/LibreDWG/libredwg/files/10717187/PL2.scr.gz)
[PL2.DXF.gz](https://github.com/LibreDWG/libredwg/files/10717188/PL2.DXF.gz)
[PL2.DWG.gz](https://github.com/LibreDWG/libredwg/files/10717189/PL2.DWG.gz)

Output:
```
type: 18 [RCd]
Add entity POLYLINE_2D [14] Decode entity POLYLINE_2D
===========================
Entity number: 14, Type: 18, Addr: 6d9
flag_r11: 0x0 [RC 0]
size: 26 [RS]
layer: 0x0 [H(RSx) 8]
opts_r11: 0x8000 [RSx 0]
          HAS_EXTRA(0x8000)
extra1_r11: 0xdf8e [RLL 0]
extra2_r11: 0x0 [RLL 0]
crc: CBD0 [RSx]
Warning: check_CRC mismatch 1753-1777 = 24: 0000 <=> CBD0
```

Another case is:
https://github.com/michal-josef-spacek/CAD-Format-DWG-AC1009/issues/2#issuecomment-1415184077

leshasoft, 2023-02-13:
Hm. Interesting Entities:
1 byte - entities identifier
1 byte - entities property
3 bytes - offset in Extra Entities Table
1 byte  - value 0x80
2 bytes - crc16
(optional, variable) - unknown data

    0006D9 12 00    unknown, entprop 0H (0)
    0006DB 1A 00    length 1AH (26)
    0006DD 00 00 00 80
    0006E1 <8E DF> 00 00 00 00 00 00-00 00 00 00 00 00 00 00   ................
    0006F1 D0 CB

rurban, 2023-02-13:
So the r11 layer with opts 0x8000 turns into the extras offset.

rurban, 2023-02-13:
TODO: the preR13 output iterators need to be special-cased for IN_EXTRA, esp. for out_dxf, ps and svg.
out_json should stay close asis.

closed 2023-02-13 by rurban