LibreDWG/libredwg #549: Issue with LOG_FLAG_R11_MAX

michal-josef-spacek, 2023-01-03:
Code:
```
#define LOG_FLAG_R11_MAX(w)                             \
            if (_obj->flag_r11 > FLAG_R11_##w)          \
              LOG_WARN ("Unknown flag_r11 (0x%d)", _obj->flag_r11)
```

is not working.
_obj->flag_r11 is e.g. 0xa0 (160) = composition of HAS_HANDLING (0x32) and HAS_ATTRIBS (0x128)
And it's greater than the defined max value every time.

michal-josef-spacek, 2023-01-03:
[3DMESH1.DWG.gz](https://github.com/LibreDWG/libredwg/files/10337736/3DMESH1.DWG.gz)

Output:
```
type: 19 [RCd]
Add entity POLYLINE_MESH [13] Decode entity POLYLINE_MESH
===========================
Entity number: 13, Type: 19, Addr: 6cf
flag_r11: 0xa0 [RC 0]
          HAS_HANDLING (0x32) HAS_ATTRIBS (0x128) Warning: Unknown flag_r11 (0x160)
  
size: 23 [RS]
layer: 0 [H(RSd) 8]
opts_r11: 0x1f1 [RSx 0]
          HAS_FLAG (1) HAS_M_VERTS (16) HAS_N_VERTS (32) HAS_M_DENSITY (64) HAS_N_DENSITY (128) HAS_CURVETYPE (256) Warning: Unknown opts_r11 (0x497)

handle: 0.1.26 [H 5]
flag: 0x14 [RC 70]
      SPLINE_FIT (4) MESH (16)
num_m_verts: 3 [RS 71]
num_n_verts: 3 [RS 72]
m_density: 7 [RS 73]
n_density: 7 [RS 74]
curve_type: 0x5 [RC 75]
ERROR: offset 1
crc: AA50 [RSx]
```

michal-josef-spacek, 2023-01-03:
Fix: https://github.com/LibreDWG/libredwg/pull/554

rurban, 2023-01-04:
Thanks, fixed

closed 2023-01-04 by rurban