LibreDWG/libredwg #558: issue with tmp line entity

michal-josef-spacek, 2023-01-04:
Files:
[tmp_line.scr.gz](https://github.com/LibreDWG/libredwg/files/10347712/tmp_line.scr.gz)
[tmp_line.dxf.gz](https://github.com/LibreDWG/libredwg/files/10347713/tmp_line.dxf.gz)
[tmp_line.dwg.gz](https://github.com/LibreDWG/libredwg/files/10347714/tmp_line.dwg.gz)

There is tmp line entity actually:
```
type: 129 [RCd]
Add entity LINE [11] Decode entity LINE
===========================
Entity number: 11, Type: 129, Addr: 512
flag_r11: 0x4 [RC 0]
          HAS_ELEVATION (4)
size: 40 [RS]
layer: 0 [H(RSd) 8]
opts_r11: 0x0 [RSx 0]
elevation_r11: 6.000000 [RD 38]
start: (1.000000, 7.000000) [2RD 10]
end: (2.000000, 0.000000) [2RD 11]
ERROR: offset -8
```
Type: 129 → bad usage of elevation

michal-josef-spacek, 2023-01-05:
See difference between line and tmp line in R10 output:

[tmp_line2.scr.gz](https://github.com/LibreDWG/libredwg/files/10350702/tmp_line2.scr.gz)
[tmp_line2.dxf.gz](https://github.com/LibreDWG/libredwg/files/10350703/tmp_line2.dxf.gz)
[tmp_line2.dwg.gz](https://github.com/LibreDWG/libredwg/files/10350704/tmp_line2.dwg.gz)

```
entities: (0x512-0x59c (0), size 138, 0x0)
==========================================
type: 1 [RCd]
Add entity LINE [11] Decode entity LINE
===========================
Entity number: 11, Type: 1, Addr: 512
flag_r11: 0x4 [RC 0]
          HAS_ELEVATION (4)
size: 40 [RS]
layer: 0 [H(RSd) 8]
opts_r11: 0x0 [RSx 0]
start: (1.000000, 1.000000) [2RD 10]
end: (2.000000, 2.000000) [2RD 11]

type: 129 [RCd]
Add entity LINE [12] Decode entity LINE
===========================
Entity number: 12, Type: 129, Addr: 53a
flag_r11: 0x4 [RC 0]
          HAS_ELEVATION (4)
size: 40 [RS]
layer: 0 [H(RSd) 8]
opts_r11: 0x0 [RSx 0]
elevation_r11: 6.000000 [RD 38]
start: (1.000000, 7.000000) [2RD 10]
end: (2.000000, 0.000000) [2RD 11]
ERROR: offset -8
```

michal-josef-spacek, 2023-01-05:
This https://github.com/LibreDWG/libredwg/blob/master/src/common_entity_data.spec#L182 ?

rurban, 2023-01-05:
> This https://github.com/LibreDWG/libredwg/blob/master/src/common_entity_data.spec#L182 ?

Ah, indeed.  Wonder I got that from. Looks wrong.

rurban, 2023-01-05:
I got it now why. Deleted entities don't match this obj->type filter. I rather should use fixedtype

closed 2023-01-05 by rurban