LibreDWG/libredwg #584: supported .dwg format versions?

abubelinha, 2023-01-18:
Not a bug, but a question.
I am not a CAD user, but I've got some .dwg files that I need to read. 

Looking for a program which can convert .dwg files produced by Autocad 2022 to some other format which I can read.
I suppose they are just the default Autocad 2022 output format.  
Is libredwg supposed to handle that, or is it limited to older versions? 

Thanks and sorry about the newbie question.

fmunozs, 2023-01-18:
Hello, 

According to [1] Autocad 2022 should use 2018 format and I see some included examples on the test data with that version, so I guess its supported.

My recommendation is to compile and test it! there is an utility called dwg2svg included on libredwg that could help, if it doesnt work you could share some sample files to this project.

Also, have you tried importing them on librecad? it uses a different lib for reading DWG, not sure about the support. 

Regards

[1] https://knowledge.autodesk.com/support/autocad/learn-explore/caas/sfdcarticles/sfdcarticles/AutoCAD-drawing-file-format.html

rurban, 2023-01-19:
LibreDWG can read all DWG format versions, also 2022. If not, I'd like an example

abubelinha, 2023-01-19:
> You can use the Discussion to get this kind of help! https://github.com/LibreDWG/libredwg/discussions

Sorry, I wasn't aware of that. Good to know, thanks!

> LibreDWG can read all DWG format versions, also 2022. If not, I'd like an example

Great. I'll try LibreDWG to see if it works. 
As it's a private map I can't post those files here: how can I share them privately @rurban ?

Anyway, as @fmunozs suggests I posted a couple of questions in Discussion https://github.com/LibreDWG/libredwg/discussions/585
Thank you both for your help.

rurban, 2023-01-19:
private sharing is ok, sure.

closed 2023-01-19 by rurban