LibreDWG/libredwg #586: r10 LINE missing 2 doubles

rurban, 2023-01-19:
Some lines do have in the very same DWG, e.g. TUTORIAL.DWG shipped with AutoCAD, no missing RD's, and some miss 2.

```
Add entity LINE [15] Decode entity LINE
===========================
Entity number: 15, Type: 1, Addr: 5b2
flag_r11: 0x4 [RC 0]
          HAS_ELEVATION (4) 
size: 40 [RS]
layer: 5 [H(RSd) 8]
opts_r11: 0x0 [RSx 0]
start: (85.226573, 183.698287) [2RD 10]
end: (117.226573, 154.698287) [2RD 11]

type: 1 [RCd]
Add entity LINE [81] Decode entity LINE
===========================
Entity number: 81, Type: 1, Addr: 135a
flag_r11: 0x0 [RC 0]
size: 56 [RS]
layer: 6 [H(RSd) 8]
opts_r11: 0x0 [RSx 0]
start: (20.000000, 51.000000, -30.000000) [3RD 10]
end: (20.000000, -51.000000, -30.000000) [3RD 11]

type: 1 [RCd]
Add entity LINE [87] Decode entity LINE
===========================
Entity number: 87, Type: 1, Addr: 1482
flag_r11: 0x4 [RC 0]
          HAS_ELEVATION (4) 
size: 56 [RS]
layer: 6 [H(RSd) 8]
opts_r11: 0x0 [RSx 0]
start: (20.000000, 51.000000) [2RD 10]
end: (20.000000, -51.000000) [2RD 11]
ERROR: offset 16

```

Maybe we should look at the size also.

michal-josef-spacek, 2023-01-19:
In this case it's z coordinate -10. This is output of DXF file.
```
  0
LINE
  8
MODEL
  6
BYBLOCK
 62
     0
 10
-24.7
 20
29.0
 30
-10.0
 11
-19.3
 21
29.0
 31
-10.0
  0
LINE
  8
MODEL
 10
0.0
 20
51.0
 30
-10.0
 11
0.0
 21
-51.0
 31
-10.0
```

michal-josef-spacek, 2023-01-19:
Probably bug of AutoCAD

michal-josef-spacek, 2023-01-19:
Ah sorry, there is no move of variables. So What's result of this values?

rurban, 2023-01-19:
No idea what they mean. Very much a bug

closed 2023-01-20 by rurban