LibreDWG/libredwg #587: How to name DIMENSION_ANG3PT DXF16 variable?

michal-josef-spacek, 2023-01-19:
Example:
[ANG3PT.scr.gz](https://github.com/LibreDWG/libredwg/files/10457003/ANG3PT.scr.gz)
[ANG3PT.DXF.gz](https://github.com/LibreDWG/libredwg/files/10457004/ANG3PT.DXF.gz)
[ANG3PT.DWG.gz](https://github.com/LibreDWG/libredwg/files/10457005/ANG3PT.DWG.gz)

[image]

There is DXF16 variable in DWG file, but R14 dblist don't show it.
How to name it?

michal-josef-spacek, 2023-01-19:
btw: Variable is not in DXF file.

rurban, 2023-01-19:
That would be the def_pt

michal-josef-spacek, 2023-01-19:
def_pt is here.
In DXF doc is:
```
16 Point defining dimension arc for angular dimensions
```

rurban, 2023-01-19:
Maybe this:
```
    if (R11OPTS (64)) {
      FIELD_RD0 (horiz_direction, 51);
      FIELD_RD0 (dim_rotation, 53);
    }
``` but I'm still looking for better docs

michal-josef-spacek, 2023-01-19:
I don't think that this is true. As I see, all option has one variable. Probably in this case is DXF 16 sometime used. And default is 0,0 so not in DXF.

rurban, 2023-01-19:
RD0 means that a default of 0.0 is missing from the DXF. and the DXF docs mentions those 2. And with the final dimstyle handle it works fine.

rurban, 2023-01-19:
I rather think it is now:
horiz_direction: 0.000000 [RD 51]
text_rotation: 0.000000 [RD 53]
dimstyle: 2 [H(RSd) 0]

rurban, 2023-01-19:
I also have now:
tutorial_from_autocad_r10_xenix_r10-master.log

type: 23 [RCd]
Add entity DIMENSION_ANG2LN [23] ===========================
Entity number: 23, Type: 23, Addr: 704
flag_r11: 0x0 [RC 0]
size: 114 [RS]
layer: 3 [H(RSd) 8]
opts_r11: 0x19 [RSx 0]
          UNKNOWN_1 (1) HAS_DXF13 (8) HAS_DXF14 (16) 
block: 0 [H(RSd) 2]
def_pt: (200.000000, 290.000000, 0.000000) [3RD 10]
text_midpt: (173.900000, 290.000000) [2RD 11]
=> Entity DIMENSION_LINEAR
xline1_pt: (100.000000, 0.000000, 232.000000) [3RD 13]
xline2_pt: (229.000000, 0.000000, 200.000000) [3RD 14]
ERROR: offset 16

michal-josef-spacek, 2023-01-19:
> I also have now: tutorial_from_autocad_r10_xenix_r10-master.log
> 
> flag_r11: 0x0 [RC 0] size: 114 [RS] layer: 3 [H(RSd) 8] opts_r11: 0x19 [RSx 0] UNKNOWN_1 (1) HAS_DXF13 (8) HAS_DXF14 (16) block: 0 [H(RSd) 2] def_pt: (200.000000, 290.000000, 0.000000) [3RD 10] text_midpt: (173.900000, 290.000000) [2RD 11] => Entity DIMENSION_LINEAR xline1_pt: (100.000000, 0.000000, 232.000000) [3RD 13] xline2_pt: (229.000000, 0.000000, 200.000000) [3RD 14] ERROR: offset 16
> 
> ie with opts 1 another 2RD point with LINEAR

Missing ```clone_ins_pt: (100.000000, 0.000000) [2RD 12]```
I have:
```type: 23 [RCd]
Add entity DIMENSION_ANG2LN [23] ===========================
Entity number: 23, Type: 23, Addr: 704
flag_r11: 0x0 [RC 0]
size: 114 [RS]
layer: 3 [H(RSd) 8]
opts_r11: 0x19 [RSx 0]
          HAS_DXF12 (1) HAS_DXF13 (8) HAS_DXF14 (16)
block: 0 [H(RSd) 2]
def_pt: (200.000000, 290.000000, 0.000000) [3RD 10]
text_midpt: (173.900000, 290.000000) [2RD 11]
clone_ins_pt: (100.000000, 0.000000) [2RD 12]
=> Entity DIMENSION_LINEAR
xline1_pt: (232.000000, 229.000000, 0.000000) [3RD 13]
xline2_pt: (200.000000, 200.000000, 0.000000) [3RD 14]

```

michal-josef-spacek, 2023-01-19:
Probably missing commit: https://github.com/LibreDWG/libredwg/commit/f0f7df1926bbb559923301e75553ca9b789b0023

michal-josef-spacek, 2023-01-19:
I don't understand term ```dimension arc``` in ```DXF 16: Point defining dimension arc for angular dimensions```.
Do you know what this mean in AutoCAD?

michal-josef-spacek, 2023-01-19:
Example:
[DIM.DXF.gz](https://github.com/LibreDWG/libredwg/files/10457902/DIM.DXF.gz)
[DIM.DWG.gz](https://github.com/LibreDWG/libredwg/files/10457903/DIM.DWG.gz)

My output:
```
Add entity DIMENSION_ANG2LN [13] ===========================
Entity number: 13, Type: 23, Addr: 6cf
flag_r11: 0x20 [RC 0]
          HAS_HANDLING (32)
size: 153 [RS]
layer: 0 [H(RSd) 8]
opts_r11: 0x847a [RSx 0]
          HAS_FLAG (2) HAS_DXF13 (8) HAS_DXF14 (16) HAS_DXF15 (32) HAS_DXF16 (64) HAS_DXF53 (1024) HAS_DIMSTYLE (32768)
handle: 0.1.5B [H 5]
block: 0 [H(RSd) 2]
def_pt: (7.453637, 6.190869, 0.000000) [3RD 10]
text_midpt: (7.868650, 5.811528) [2RD 11]
flag: 0x85 [RC 70]
=> Entity DIMENSION_ANG3PT
xline1_pt: (6.060436, 6.406088, 0.000000) [3RD 13]
xline2_pt: (8.708595, 5.485104, 0.000000) [3RD 14]
center_pt: (9.208638, 7.694218, 0.000000) [3RD 15]
xline2end_pt: (0.000000, 0.000000) [2RD 16]
text_rotation: 0.209440 [RD 53]
dimstyle: 0 [H(RSd) 0]
crc: 84D5 [RSx]
```
With https://github.com/LibreDWG/libredwg/pull/588
and change in DIMENSION_ANG3PT:
```
+        if (R11OPTS (64))
+          FIELD_2RD (xline2end_pt, 16);
```

rurban, 2023-01-20:
FIELD_2RD (xline2end_pt, 0) really. there's no DXF output. And it could still be some angles

rurban, 2023-01-20:
`smoke/dim` has all these intergrated now.

closed 2023-01-20 by rurban