LibreDWG/libredwg #589: Issue with block_start in case with many linetypes

michal-josef-spacek, 2023-01-20:
Example:
[LT.scr.gz](https://github.com/LibreDWG/libredwg/files/10468962/LT.scr.gz)
[LT.DWG.gz](https://github.com/LibreDWG/libredwg/files/10468963/LT.DWG.gz)

```
entities_start: 0x2e5 [RLx 0]
entities_end: 0x2e5 [RLx 0]
blocks_start: 0xffffc0bd [RLx 0]
blocks_size: 0x0 [RLx 0]
blocks_end: 0x0 [RLx 0]
blocks_max: 0x80000000 [RLx 0]
```
and
```contents table VIEW     [ 6]: size:91   num:0   (0xffffc0bd-0xffffc0bd)
Warning:
@0xc0bd => blocks_start 0xffffc0bd
ERROR: bit_read_TF buffer overflow at 49341.4294901760 + 0 > 49341
unknown (-65536):
```

rurban, 2023-01-21:
Actually I thought about that also, what the high bit and byte could mean. Thanks

rurban, 2023-01-25:
Fixed

closed 2023-01-25 by rurban

michal-josef-spacek, 2023-01-25:
Thanks, working