LibreDWG/libredwg #591: Fix num_sections in decode_r11.c

michal-josef-spacek, 2023-01-20:
There is an init value (https://github.com/LibreDWG/libredwg/blob/master/src/decode_r11.c#L708):
```
unsigned num_sections = 5;
```

But not adding sections like (https://github.com/LibreDWG/libredwg/blob/master/src/dwg.c#L3051):
```
if (!dwg->header.numsections)
        dwg->header.numsections = 5;
      // HEADER.numsections is always 5 even if it needs to be 8 or 10,
      // probably because the additional sections are embedded in HEADER_VARS.
      // 5 tables + header + block. VIEW = 6
      if (dwg->header.numheader_vars > 158) // r10
        num_sections += 3;
      if (dwg->header.numheader_vars > 160) // r11
        num_sections += 2;
      // and there is one hole we need to skip over.
      num_sections += 1;
}
```

The actual situation is, that sections are not decoding.

rurban, 2023-01-26:
I cannot repro that. The sections depend on numheader_vars (not version)

michal-josef-spacek, 2023-01-26:
Main issue is parsing of sections from R10.

Example:
[UCS.DXF.gz](https://github.com/LibreDWG/libredwg/files/10507752/UCS.DXF.gz)
[UCS.DWG.gz](https://github.com/LibreDWG/libredwg/files/10507753/UCS.DWG.gz)

In example are sections:
layer, style, ltype, view, ucs and vport
In dwgread debug is only:
```
contents table VIEW     [ 6]: size:149  num:0   (0x6b4-0x6b4)
Warning:
@0x6b4 => blocks_start 0x90f
...
```

And I think, that's because num_sections isn't 7, but only 5

rurban, 2023-01-27:
Oops, I broke it.

rurban, 2023-01-27:
See 139e100f6052cd9be519a048998504ff5438784d

closed 2023-01-30 by rurban