LibreDWG/libredwg #592: Unify unknown variable in header (preR13 and R13 versions)

michal-josef-spacek, 2023-01-20:
We have different variables for same thing in preR13 and R13 versions

Examples:
[Drawing_r14.dxf.gz](https://github.com/LibreDWG/libredwg/files/10469189/Drawing_r14.dxf.gz)
[Drawing_r14.dwg.gz](https://github.com/LibreDWG/libredwg/files/10469190/Drawing_r14.dwg.gz)
[Drawing_r11.dxf.gz](https://github.com/LibreDWG/libredwg/files/10469191/Drawing_r11.dxf.gz)
[Drawing_r11.dwg.gz](https://github.com/LibreDWG/libredwg/files/10469192/Drawing_r11.dwg.gz)

In R11:
```
UNITMODE: 0 [RS 70]
DEBUG_POS @1341.0 (10728)
000000001C007EFD5742000000000000F03F000000000000
  RS :0 / 0x0000 (16)

  RL :0 / 0x00000000 (32)
  RD :-3065654374452100702700068470377681718323237861525184809381632932604485779920863438634634480952183588279191327718704389405575126647253555972756696551548588542298848213776963404130111761
27619814375290017650179825415209347897054436268637303863604519312689688318123099918712607357526589156302444822528.000000 (64)
unknown (34): 000000001C007EFD5742000000000000F03F000000000000F03F000000000000F03F
unknown_unit1: "m" [TFv 32 1]
unknown_unit2: "" [TFv 32 1]
unknown_unit3: "" [TFv 32 1]
unknown_unit4: "" [TFv 32 1]
```

In R14:
```
unknown_0: 412148564080.000000 [BD 0]
unknown_1: 1.000000 [BD 0]
unknown_2: 1.000000 [BD 0]
unknown_3: 1.000000 [BD 0]
unknown_text1: "m" [TV 0]
unknown_text2: "" [TV 0]
unknown_text3: "" [TV 0]
unknown_text4: "" [TV 0]
```

In R11 are 412148564080, 1, 1, 1 before unknown_unit1 too.

rurban, 2023-01-21:
Actually their meaning is known. 4 conversion unit ratios and its names.

michal-josef-spacek, 2023-01-21:
> Actually their meaning is known. 4 conversion unit ratios and its names.

Ok. Have you got an example of how is it used in real AutoCAD? I don't know.

michal-josef-spacek, 2023-01-22:
https://github.com/LibreDWG/libredwg/pull/599

rurban, 2023-01-25:
Merged

closed 2023-01-25 by rurban