LibreDWG/libredwg #601: Fix make regen-dynapi on Debian

michal-josef-spacek, 2023-01-23:
Actuall situation in my Debian:
```
/usr/bin/perl ./src/gen-dynapi.pl
using gcc with /usr/lib/gcc/x86_64-linux-gnu/10/include
stdint.h, line 26: file 'bits/libc-header-start.h' not found
        included from /usr/lib/gcc/x86_64-linux-gnu/10/include/stdint.h:9
        included from ./include/dwg.h:33 at ./src/gen-dynapi.pl line 83.
make: [Makefile:1259: regen-dynapi] Chyba 2 (ignorováno)
```

Output of ```gcc -print-search-dirs``` is:
```
install: /usr/lib/gcc/x86_64-linux-gnu/10/
programs: =/usr/lib/gcc/x86_64-linux-gnu/10/:/usr/lib/gcc/x86_64-linux-gnu/10/:/usr/lib/gcc/x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/10/:/usr/lib/gcc/x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/10/../../../../x86_64-linux-gnu/bin/x86_64-linux-gnu/10/:/usr/lib/gcc/x86_64-linux-gnu/10/../../../../x86_64-linux-gnu/bin/x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/10/../../../../x86_64-linux-gnu/bin/
libraries: =/usr/lib/gcc/x86_64-linux-gnu/10/:/usr/lib/gcc/x86_64-linux-gnu/10/../../../../x86_64-linux-gnu/lib/x86_64-linux-gnu/10/:/usr/lib/gcc/x86_64-linux-gnu/10/../../../../x86_64-linux-gnu/lib/x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/10/../../../../x86_64-linux-gnu/lib/../lib/:/usr/lib/gcc/x86_64-linux-gnu/10/../../../x86_64-linux-gnu/10/:/usr/lib/gcc/x86_64-linux-gnu/10/../../../x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/10/../../../../lib/:/lib/x86_64-linux-gnu/10/:/lib/x86_64-linux-gnu/:/lib/../lib/:/usr/lib/x86_64-linux-gnu/10/:/usr/lib/x86_64-linux-gnu/:/usr/lib/../lib/:/usr/lib/gcc/x86_64-linux-gnu/10/../../../../x86_64-linux-gnu/lib/:/usr/lib/gcc/x86_64-linux-gnu/10/../../../:/lib/:/usr/lib/
```

In ```@ccincdir``` is ```/usr/lib/gcc/x86_64-linux-gnu/10/include``` which is not contain ```bits/libc-header-start.h```
Fix is something like this:
```
- ->Include('.', @ccincdir, '/usr/include/')
+ ->Include('.', @ccincdir, '/usr/include/', '/usr/include/x86_64-linux-gnu')
```

@rurban Is this fix ok, or is this something complex?

rurban, 2023-01-24:
That's too arch-specific.
Works with all my compilers on Fedora, Ubuntu and macOS

michal-josef-spacek, 2023-01-24:
@rurban yes, we need to fix with /usr/include/$ARCH

Code before this patch is ok and not a problem.
A problem is only with this arch include dir.

michal-josef-spacek, 2023-01-24:
https://github.com/LibreDWG/libredwg/pull/604

rurban, 2023-01-25:
Merged