LibreDWG/libredwg #607: Issue with logging in dwg.c

michal-josef-spacek, 2023-01-27:
There are no logging messages when I compile code which is using dwg.c.
Logging from encode.c is on the place.

When I am looking to .c file, they are similar.
Both have:
```
/* The logging level per .o */
static unsigned int loglevel; 
#ifdef USE_TRACING 
/* This flag means we have checked the environment variable
   LIBREDWG_TRACE and set `loglevel' appropriately.  */
static bool env_var_checked_p;
#endif /* USE_TRACING */
#define DWG_LOGLEVEL loglevel
#include "logging.h"
```
I don't know what happened.

Example:
[ex9.c.gz](https://github.com/LibreDWG/libredwg/files/10518407/ex9.c.gz)
Compile by:
```
gcc -std=c99 -Wall `pkg-config --cflags libredwg` -o ex9 ex9.c `pkg-config --libs libredwg`
```

rurban, 2023-01-27:
the loglevel must be passed via dat or dwg to each function

michal-josef-spacek, 2023-01-27:
Ah i see, ok thanks.

closed 2023-01-27 by michal-josef-spacek