LibreDWG/libredwg #610: dwgadd issue with parsing of entity field (entity.field)

michal-josef-spacek, 2023-01-28:
When I have dwgadd input:
```
text "baz" (1 2 3) 8
text.oblique_angle = 45.0
```

Parsing of entity not working.
Code of:
```
#define SET_ENT(var, name)                                                    \
  if (SSCANF_S (p, #var "." FMT_NAME " = %d.%d.%X", &s1[0] SZ, &i1, &i2, &u)) \
    {                                                                         \
      BITCODE_H hdl;                                                          \
      if (!ent.u.var || ent.type != DWG_TYPE_##name)                          \
        fn_error ("Invalid type " #var ". Empty or wrong type\n");            \
      hdl = dwg_add_handleref (dwg, i1, u, NULL);                             \
      dwg_dynapi_entity_set_value (ent.u.var, #name, s1, hdl, 0);             \
    }
...
```
which is for parsing of the handle will be used instead of the right parse of float.

rurban, 2023-02-01:
Wrong usage of sscanf, we must check the result of sscanf also.

rurban, 2023-02-02:
Working on it in the branch smoke/add_angles

closed 2023-02-02 by rurban