LibreDWG/libredwg #619: Fix attdef and attrib concept

michal-josef-spacek, 2023-01-31:
There is ```dwg_add_ATTRIB```, ```dwg_add_ATTDEF``` internal functions and public ```dwg_add_Atribute``` which works on insert.
This is not ok.

```ATTDEF``` is a regular entity that could be in entities as is.
```ATTDEF``` entity could be used in block, this is a sense of this entity. In this case, it is moving to block (between ```BLOCK``` and ```ENDBLK``` entities).
```ATTRIB``` is an entity which is every time in insert sequence (between ```INSERT``` and ```SEQEND``` entities).

michal-josef-spacek, 2023-01-31:
WIP for this issue: https://github.com/LibreDWG/libredwg/pull/613

michal-josef-spacek, 2023-01-31:
@rurban 
Be carefull with https://github.com/LibreDWG/libredwg/commit/bf6701310bba9d573bc648a574b74429eaa2ec97#diff-e54aa6e0bc5912a840bf5dbc5bb4c63d6a2453d58f60b30287b1e400d3702451R12
```
attrib.text = "text1"
```
There is possibility of:
```
block "bloko"
attdef 8 0 "prompt" (0.0 1.0 0.0) "tag" "default_text"
line (0 1 0) (1 1 0)
attdef 8 0 "prompt" (1.0 2.0 0.0) "tag" "default_text"
endblk
```
Same tag and insert will be with two texts.

rurban, 2023-01-31:
Hmm, then we'd need an optional attribN support

rurban, 2023-02-01:
More importantly:
Our add API resembles the VBA API. There's a special AddAttribute, and a AttributeReference.

The VBA reference names it Attribute (the ATTDEF) and AttributeReference (the ATTRIB)
http://entercad.ru/acadauto.en/idh_attribute_object.htm

AttributeReference (ATTRIB) cannot be created via Add, only via InsertBlock, (our add_INSERT).
So we need to insert ATTRIB's automatically with some add API, maybe dwg_InsertBlock.

rurban, 2023-02-01:
> Hmm, then we'd need an optional attribN support

No, I'd set the individual attrib fields one after the other. Implemented in `smoke/export_attdef`

michal-josef-spacek, 2023-02-01:
> More importantly: Our add API resembles the VBA API. There's a special AddAttribute, and a AttributeReference.
>
> The VBA reference names it Attribute (the ATTDEF) and AttributeReference (the ATTRIB) [...]

it seems reasonable

rurban, 2023-02-01:
add_ATTRIB and add_ATTDEF now properly implemented, with dwgadd examples.
Just the automatic InsertBlock not. I'll probably wont do that

closed 2023-02-01 by rurban