LibreDWG/libredwg #628: dwgadd: Fix attdef and attrib tag for releases which could use utf8

michal-josef-spacek, 2023-02-02:
attdef and atrib tag is one word and R13 (tested on R14) supports utf8 string for it.
dwgadd don't encode utf8 string in case of tag actually.

rurban, 2023-02-03:
Fixed by f6db2bd4a65317348dddbc56e313079063a36c08 ?

michal-josef-spacek, 2023-02-03:
> Fixed by [f6db2bd](https://github.com/LibreDWG/libredwg/commit/f6db2bd4a65317348dddbc56e313079063a36c08) ?

No. There is FMT_TBL for tag and this don't support utf8

michal-josef-spacek, 2023-02-03:
btw: I don't know how about encodings in preR13 :-D

rurban, 2023-02-03:
The preR13 encodings are single-byte system-specific, but our dwg_add API accepts utf-8 only and converts its to the used encoding.

michal-josef-spacek, 2023-02-03:
I don't think that could be FMT_ANY, tag could be only one word

rurban, 2023-02-04:
So I actuallly did read the docs now. You are right. They are special.

rurban, 2023-02-06:
Fixed

closed 2023-02-06 by rurban