LibreDWG/libredwg #632: Fix AUX header in R11

michal-josef-spacek, 2023-02-08:
We need to fix AUX header in R11

Prepared commit: https://github.com/michal-josef-spacek/libredwg/commit/12699be71a2f47422e3adef72ff81c8ed6d25405
Possible to commit after https://github.com/LibreDWG/libredwg/pull/630 and https://github.com/LibreDWG/libredwg/pull/631

michal-josef-spacek, 2023-02-08:
Only thing which i don't know is crc16

michal-josef-spacek, 2023-02-08:
Example output:
```
AUXHEADER: @0x1635e
DWG_SENTINEL_R11_AUX_HEADER_BEGIN: 298DD149A9731FEA99DE32F94D0AE019
unknown: 16 [RS 0]
auxheader_size: 138 [RS 0]
entities_start: 0x6cf [RLx 0]
entities_end: 0x1525c [RLx 0]
block_entities_start: 0x162c8 [RLx 0]
extra_entities_start: 0x1634e [RLx 0]
R11_HANDLING: 0 [RS 0]
R11_HANDSEED: 0.8.8 [H 5]
num_aux_tables: 10 [RS 0]
chk table BLOCK    [ 1]: size:45   nr:1   (0x1527c)
chk table LAYER    [ 2]: size:41   nr:10  (0x15404)
chk table STYLE    [ 3]: size:198  nr:2   (0x155be)
chk table LTYPE    [ 5]: size:191  nr:1   (0x15c0e)
chk table VIEW     [ 6]: size:153  nr:0   (0x15ced)
chk table UCS      [ 7]: size:109  nr:0   (0x15d0d)
chk table VPORT    [ 8]: size:253  nr:4   (0x15d2d)
chk table APPID    [ 9]: size:37   nr:1   (0x16141)
chk table DIMSTYLE [10]: size:324  nr:0   (0x16186)
chk table VX       [11]: size:43   nr:6   (0x161a6)
auxheader_address: 0x1635e [RLx 0]
@0x163f6
crc: 21B2 [RSx] from 0x1635e-0x163f6
ERROR: AUX header CRC mismatch 21B2 <=> 14A0
DWG_SENTINEL_R11_AUX_HEADER_END: D6722EB6568CE0156621CD06B2F51FE6
```

rurban, 2023-02-08:
I just dont like the block_entities_start name, extra ok, but please keep the old name

michal-josef-spacek, 2023-02-08:
@rurban I reflect name in header, which i changed :-)
Feel free to change.

leshasoft, 2023-02-09:
There, the checksum is considered without the upper sentinel.

rurban, 2023-02-09:
I could have told you that :)
But still busy with your PR's

leshasoft, 2023-02-09:
> AUXHEADER: @0x1635e
> DWG_SENTINEL_R11_AUX_HEADER_BEGIN: 298DD149A9731FEA99DE32F94D0AE019

....

> crc: 21B2 [RSx] from 0x1635e-0x163f6

It was not a question.

michal-josef-spacek, 2023-02-09:
I tested calculating of crc after upper sentinel, but without success. 
The question is if seed (0xC0C1) is valid. Or some bug in the code.

leshasoft, 2023-02-09:
I tried on several files, I got the equal checksum, the seed (0xC0C1) is valid.

rurban, 2023-02-09:
You can rebase it now on master

michal-josef-spacek, 2023-02-09:
PR prepared: https://github.com/LibreDWG/libredwg/pull/633

closed 2023-02-09 by rurban