LibreDWG/libredwg #635: Fix extra entities section in preR13

michal-josef-spacek, 2023-02-09:
We need to fix the extra entity section. This is not a classic entity section.

Example:
[PL1.DXF.gz](https://github.com/LibreDWG/libredwg/files/10702438/PL1.DXF.gz)
[PL1.DWG.gz](https://github.com/LibreDWG/libredwg/files/10702439/PL1.DWG.gz)

Actual output:
```
extra entities: (0x932-0x943 (0), size 17)
==========================================
type: 19 [RCd]
Add entity POLYLINE_2D [23] Decode entity POLYLINE_2D
===========================
Entity number: 23, Type: 19, Addr: 932
flag_r11: 0x80 [RC 0]
          HAS_ATTRIBS(0x80)
size: 9 [RS]
layer: 0 [H(RSd) 8]
opts_r11: 0x1 [RSx 0]
          HAS_FLAG(0x1)
flag: 0x1 [RC 70]
      CLOSED(0x1)

type: 18 [RCd]
Add entity POLYLINE_2D [24] Decode entity POLYLINE_2D
===========================
Entity number: 24, Type: 18, Addr: 93b
flag_r11: 0x0 [RC 0]
size: 8 [RS]
layer: 1306 [H(RSd) 8]
opts_r11: 0x0 [RSx 0]
==========================================
extra entities: end
```

The second item isn't an entity, but extra information for replacement. Every time is 8 bytes in length.

rurban, 2023-02-10:
Seem to be the curve fit pts. Later they are calculated on the fly, but with the old machines they might have stored it there. DXF and rendering needs it.

rurban, 2023-02-13:
This seems to be fixed now, just the polylines using it not yet.
Just that the layer is really an offset

rurban, 2023-02-13:
TODO: the preR13 output iterators need to be special-cased for IN_EXTRA, esp. for out_dxf, ps and svg.
out_json should stay close asis.

michal-josef-spacek, 2023-02-28:
Related issue: https://github.com/michal-josef-spacek/CAD-Format-DWG-AC1009/issues/4

michal-josef-spacek, 2023-03-21:
I created PR for jump entity: https://github.com/LibreDWG/libredwg/pull/670

example output:
```
entities: (0x6cf-0x735 (0), size 102)
==========================================
DWG_SENTINEL_R11_ENTITIES_BEGIN: C46E6854F86E3330633EC1852ADC9401
@offset: 0x0
type: 18 [RCd]
Add entity JUMP [13] Decode entity JUMP
===========================
Entity number: 13, Type: 18, Addr: 6cf
flag_r11: 0x0 [RC 0]
size: 10 [RS]
jump_address_raw: 0x80000187 [RLx 0]
jump_entity_section: DWG_EXTRA_SECTION
jump_address: 0x187
crc: 3BF5 [RSx]
 check_CRC 1743-1751 = 8: 3BF5 == 3BF5

@offset: 0xa
type: 18 [RCd]
Add entity JUMP [14] Decode entity JUMP
===========================
Entity number: 14, Type: 18, Addr: 6d9
flag_r11: 0x0 [RC 0]
size: 26 [RS]
jump_address_raw: 0x80000000 [RLx 0]
jump_entity_section: DWG_EXTRA_SECTION
jump_address: 0x0
crc: DF8E [RSx]
 check_CRC 1753-1761 = 8: DF8E == DF8E
trailing (16): 0000000000000000000000000000D0CB
...
```

rurban, 2023-04-08:
Added documentation into the official texi and here: https://github.com/LibreDWG/libredwg/discussions/691

rurban, 2023-04-18:
The iterator now honors JUMP.
DXF does not work yet due to wrong BLOCK_HEADER.entities[] but this is something different

closed 2023-04-18 by rurban