LibreDWG/libredwg #639: BUG of cvt_TIMEBLL

gongxy-sh, 2023-02-13:
-		_obj->TDCREATE	{days=2454762 ms=99310234 value=2454762.9931023400 }	_dwg_time_bll
		days	2454762	unsigned int
		ms	99310234	unsigned int
		value	2454762.9931023400	double
-		tm	{tm_sec=10 tm_min=35 tm_hour=27 ...}	tm
		tm_sec	10	int
		tm_min	35	int
		tm_hour	27	int
		tm_mday	22	int
		tm_mon	9	int
		tm_year	108	int
tm_hour = 27 is invalid

rurban, 2023-02-13:
Thanks. I think that deserves a proper unit test, to check for invalid values

gongxy-sh, 2023-02-13:
The problem is that the maximum value of ms is 24 * 60 * 60 * 1000=86400000, but from example_ The first time value read by 2000.dwg (dwg_read_file) is 99310234 > 86400000, which is incorrect.

gongxy-sh, 2023-02-13:
The problem is that the maximum value of ms is 24 * 60 * 60 * 1000=86400000, the first time value read by example_2000.dwg (dwg_read_file) is 99310234 > 86400000, which is incorrect.

gongxy-sh, 2023-02-13:
According to value=days+(ms * 1e-8), I think the maximum ms is not 86400000, but 1E8, so the number of revolutions per millisecond needs to be multiplied by 0.864.

gongxy-sh, 2023-02-13:
t = date.ms * 0.864 / 1000.0; /* in seconds */

rurban, 2023-02-13:
> According to value=days+(ms * 1e-8), I think the maximum ms is not 86400000, but 1E8, so the number of revolutions per millisecond needs to be multiplied by 0.864.

Well, the value format is just a readable format for their datetime. The real max. ms is what you calculated.

rurban, 2023-02-17:
merged thanks

closed 2023-02-17 by rurban