LibreDWG/libredwg #646: Memory management isolation

gongxy-sh, 2023-02-15:
vs2019  link with libredwg-0.12.5.5140-win64

Dwg_Data* pDoc = dwg_new_Document(R_2000, 0, 0);
Dwg_Object* poMSpace = dwg_model_space_object(pDoc);
Dwg_Object_BLOCK_HEADER* pMSpace = poMSpace->tio.object->tio.BLOCK_HEADER;
dwg_point_3d start = { 0 };
dwg_point_3d end = { 0 };
dwg_add_LINE(pMSpace, &start, &end);
dwg_free_Document(pDoc);
dwg_free(pDoc);
free(pDoc);

will crash
remove free(pDoc); will cause memory leak.
So we need to do memory isolation ?

rurban, 2023-02-15:
It's just a double free

gongxy-sh, 2023-02-15:
//dwg_free_Document(pDoc);
dwg_free(pDoc);
free(pDoc);

dwg_free_Document not exist in  libredwg-0.12.5.5140-win64

gongxy-sh, 2023-02-15:
dwgread.c

main()
...

Dwg_Data dwg;
...
memset (&dwg, 0, sizeof (Dwg_Data));
...
dwg_read_file(..., &dwg);
...
dwg_write_xxx (, &dwg);
...
dwg_free (&dwg);

so dwg_free can't free *dwg if dwg is malloced/calloced.

gongxy-sh, 2023-02-15:
dwg_api.c
...
EXPORT Dwg_Data *
dwg_new_Document (const Dwg_Version_Type version, const int imperial,
                  const int log_level)
{
  Dwg_Data *dwg = calloc (1, sizeof (Dwg_Data));

closed 2023-02-17 by gongxy-sh