LibreDWG/libredwg #649: configure failed in centos 7.0

yangxiaomin08, 2023-02-28:
hi, I tried to compile the lib in centos 7.0, unfortunately it failed. The error message is below, can anyone have any idea about it? Thanks. 

$sh ./autogen.sh
+ rm -rf autom4te.cache config.status
+ test -d .git -a -f build-aux/git-version-gen
+ git fetch --tags
+ rm .tarball-version
++ build-aux/git-version-gen .tarball-version
+ v=0.12.5.5135_dirty
+ echo 0.12.5.5135_dirty
+ echo 0.12.5.5135_dirty
+ set -e
+ autoreconf --install --symlink -I m4
aclocal: error: aclocal: file 'm4/lt~obsolete.m4' does not exist
autoreconf: aclocal failed with exit status: 1

$automake --version
automake (GNU automake) 1.14
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl-2.0.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Tom Tromey <tromey@redhat.com>
       and Alexandre Duret-Lutz <adl@gnu.org>.

yangxiaomin08, 2023-02-28:
Is it possible that I rollback to automake 1.13?

yangxiaomin08, 2023-02-28:
update:
I have removed all the *.m4 in m4 directory, and re-run the autogen.sh. The error I met was disappeared. 

But I met again error,  then I comment the AC_DISABLE_STATIC the configure.ac, still failed as "./configure: line 13944: AX_CHECK_ALIGNED_ACCESS_REQUIRED: command not found". 

 $sh autogen.sh
 
+ autoreconf --install --symlink -I m4
configure.ac:22: error: possibly undefined macro: AC_DISABLE_STATIC
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
autoreconf: /usr/bin/autoconf failed with exit status: 1

./configure
checking dependency style of gcc... (cached) gcc3
./configure: line 13944: AX_CHECK_ALIGNED_ACCESS_REQUIRED: command not found
checking for size_t... yes
checking for int16_t... yes
checking for uint16_t... yes
checking for int32_t... yes
checking for uint32_t... yes
checking for int64_t... yes
checking for uint64_t... yes
checking size of size_t... 8
./configure: line 14090: syntax error near unexpected token `wchar_t,'
./configure: line 14090: `   AX_COMPILE_CHECK_SIZEOF(wchar_t, #include <wchar.h>)'

rurban, 2023-02-28:
this needs m4/ax_check_aligned_access_required.m4 and m4/ax_compile_check_sizeof.m4
maybe centos ships older, incompatible versions of these checks, overriding our.

install from the tarball would help then. without autogen.sh

yangxiaomin08, 2023-02-28:
Thanks a lot.  I have both tried to install from the tarball or specified the gnu99, it failed on the endian. 
As you memtioned you are working at it, when can I expect it works? thanks

rurban, 2023-02-28:
I'm a bit busy right now, but tomorrow it should be in

yangxiaomin08, 2023-02-28:
I have find a walkaround to redefine them in src/common.h, I am not familiar with libredwg yet and can't  do the test coverage. 

#define _BSD_SOURCE
#ifdef HAVE_ENDIAN_H
#  ifndef _DEFAULT_SOURCE
#    define _DEFAULT_SOURCE 1 /* for byteswap.h */
#  endif
#  include <endian.h>

#    if __BIG_ENDIAN__
#      define be64toh(n) (n)
#      define htobe64(n) (n)
#    else
#      define be64toh(n) (((uint64_t)htonl((n) & 0xFFFFFFFF) << 32) | htonl((n) >> 32))
#      define htobe64(n) (((uint64_t)htonl((n) & 0xFFFFFFFF) << 32) | htonl((n) >> 32))
#    endif /* _ENDIAN */
#elif defined HAVE_SYS_ENDIAN_H

rurban, 2023-02-28:
I've fixed now the endian problems via byteswap.h, but ran now into gcc bugs/miscompilations:
```
Reading DWG file ../test/test-data/example_2004.dwg
This file's version code is: AC1018 (r2004)
Warning: Invalid section_array_size: [25].469762048 != 28
Warning: Invalid last_section_address: 100 != 2e220
Warning: Invalid sections: 0 != numgaps: 0 + numsections: 26
Warning: section_map_id 28 not found
Warning: Add section_map_id [26] 28 => address 0x2dca0
Warning: Repair invalid section_info_id [25]: => 27
Warning: Add section_info_id [27] 27 => address 0x5b840
ERROR: bit_read_RC buffer overflow at 374848.1 + 0 > 187890
ERROR: bit_read_RS buffer overflow at 374848.0 > 187890
ERROR: bit_read_RL buffer overflow at 374848.0 > 187890
ERROR: Invalid Data Section Page Map type 0x0 != 0x4163003b
ERROR: Failed to decode file: ../test/test-data/example_2004.dwg 0x100
```

yangxiaomin08, 2023-03-01:
I have succesfully compiled with gcc7.3.0 and didn't ran onto such error "ERROR: bit_read_RC buffer overflow at 374848.1 + 0 > 187890". 

How do you run the test cases? I have tried make test but not supported and cannot find example_2000.json on the source dir.

rurban, 2023-03-01:
ERROR: bit_read_RC buffer overflow at 374848.1 + 0 > 187890" came from the wrong endian macros, fixed.
Still got crashes with the default gcc-4.8.5 on dwgwrite, not yet repro locally

rurban, 2023-03-01:
Merged. Works now for me

closed 2023-03-01 by rurban

yangxiaomin08, 2023-03-03:
Great to hear that. Thanks a lot.