LibreDWG/libredwg #652: Packaging to vcpkg?

msclock, 2023-03-02:
# Can we package the code to vcpkg?

# Are there any guidelines or permits I should follow？

rurban, 2023-03-02:
Sure you can. See the various cmake or mingw autotools recipes.

msclock, 2023-03-03:
I have submit a pr to vcpkg. https://github.com/microsoft/vcpkg/pull/30005.

This is the usage:
https://github.com/msclock/vcpkg/blob/bfa07fd074023fe0b4eb835363d280c80256572e/ports/libredwg/usage#L1-L4

rurban, 2023-03-04:
The memmem issue on Apple's 10.6 SDK is already handled by the 2nd case in common.h, just the declaration is missing. There I assume HAVE_MEMMEM is 1, but the decl in string.h is missing.

msclock, 2023-03-04:
> The memmem issue on Apple's 10.6 SDK is already handled by the 2nd case in common.h, just the declaration is missing. There I assume HAVE_MEMMEM is 1, but the decl in string.h is missing.

Well, it is. HAVE_MEMMEM was evalated as 1 but it reports `implicitly decl` error. The assumption is not valid so that I have to set an eternal true condition on Apple.

rurban, 2023-03-04:
So I'll try to come up with a better fix

rurban, 2023-08-03:
I've added now a vcpkg.json

msclock, 2023-08-04:
@rurban Got it. I've noticed that with the current quick iteration, maybe we can create our own registry for quick iterative updates, and then push it to the official VCPKG registry when it stabilizes. Please refer to:
- [create registry](https://learn.microsoft.com/en-us/vcpkg/maintainers/registries), and refer to the [example](https://github.com/northwindtraders/vcpkg-registry).
- [using registry](https://learn.microsoft.com/en-us/vcpkg/users/registries), and refer to [example](https://github.com/northwindtraders/beison-ptr).

rurban, 2023-08-04:
msclock ***@***.***> schrieb am Fr., 4. Aug. 2023, 05:01:

> @rurban <https://github.com/rurban> Got it. I've noticed that with the
> current quick iteration, maybe we can create our own registry for quick
> iterative updates, and then push it to the official [...]
I'm rather happy with you pushing to the registry for releases. Nightlies
not so much.

Reini