LibreDWG/libredwg #663: Problems with dimension arrowheads

timoria21, 2023-03-17:
Hello,

While trying to read the arrowheads of dimensions I encountered some problems.

I have created various .dwg files just for testing purposes, each one has different arrowheads:
You can find them inside the attached file [test.zip](https://github.com/LibreDWG/libredwg/files/11003424/test.zip).
For example:

- ah_dot_dot has a Dot arrowhead on the left and on the right,
- ah_dot_oblique has a Dot arrowhead on the left and an Oblique arrowhead on the right
- ...

First, if there is an Oblique arrowhead the generated .json file is **blank**.
I mean, the .json file is generated, but without any entities/tables.
Adding the -v9 argument shows many errors.

Second, looking at the DXF reference, the left arrowhead and the right arrowhead handles should be stored within the associated DIMSTYLE, in the **DIMBLK1 and DIMBLK2** fields, respectively.
But they are **empty** every time, is this normal?
(I could be wrong here, they might be stored only if an override is happening, but I'm not sure)

Thanks,

Dave

michal-josef-spacek, 2023-03-17:
@timoria21 When you look to e.g. ah_dot_dot:
There is one DIMENSION_LINEAR entity:
```
Object number: 65/41, Size: 170 [MS], Type: 21 [BS], Address: 5327
Add entity DIMENSION_LINEAR [65] Decode entity DIMENSION_LINEAR
bitsize: 1312 [RL] @3.2
 Hdlsize: 48, hdl_dat: @164.0 - @170.0 (170)
 obj string stream +1311: @163.7 (1311) has_strings: 1
 @161.7 data_size: 2/0x2 [RS]
handle: 0.1.B9 [H 5]
EED[0] size: 49 [BS]
EED[0] handle: 5.1.12
EED[0] raw: 49
00060044005300540059004C00450002004657010500000000000000C846AD004601004658010500000000000000C80201
EED[0] code: 0 [RC], wstring: len=6 [RS] "DSTYLE" [TU]
EED[1] code: 2 [RC], open: 0x0 [RC]
EED[2] code: 70 [RC], short: 343 [RS]
EED[3] code: 5 [RC], entity: 0xC800000000000000 [ulong]
EED[4] code: 70 [RC], short: 173 [RS]
EED[5] code: 70 [RC], short: 1 [RS]
EED[6] code: 70 [RC], short: 344 [RS]
EED[7] code: 5 [RC], entity: 0xC800000000000000 [ulong]
EED[8] code: 2 [RC], close: 0x1 [RC]

EED[9] size: 0 [BS] (end)
num_eed: 9
preview_exists: 0 [B 0]
entmode: 2 [BB 0]
num_reactors: 0 [BL 0]
is_xdic_missing: 1 [B 0]
color.raw: 0x100 [BSx 0]
ltype_scale: 1.000000 [BD 48]
ltype_flags: 0 [BB 0]
plotstyle_flags: 0 [BB 0]
material_flags: 0 [BB 0]
shadow_flags: 0x0 [RC 284]
invisible: 0 [BS 60]
linewt: 0x1d [RC 370]
layer: (5.1.10) abs:10 [H 8] => LAYER 0
--common_size: 502
extrusion: (0.000000, 0.000000, 1.000000) [3BD 210]
text_midpt: (43.108596, 33.529627) [2RD 11]
elevation: 0.000000 [BD 31]
flag1: 0xb [RC 0]
flag => 0x20 [RC 70]
user_text: "" [TU 1]
text_rotation: 0.000000 [BD 53]  0º
horiz_dir: 0.000000 [BD 51]  0º
ins_scale: (1.000000, 1.000000, 1.000000) [3BD_1 0]
ins_rotation: 0.000000 [BD 54]
attachment: 5 [BS 71]
lspace_style: 1 [BS 72]
lspace_factor: 1.000000 [BD 41]
act_measurement: 12.445869 [BD 42]
unknown: 0 [B 73]
flip_arrow1: 0 [B 74]
flip_arrow2: 0 [B 75]
clone_ins_pt: (0.000000, 0.000000) [2RD 12]
xline1_pt: (36.885662, 32.764791, 0.000000) [3BD 13]
xline2_pt: (49.331530, 32.764791, 0.000000) [3BD 14]
def_pt: (49.331530, 33.529627, 0.000000) [3BD 0]
oblique_angle: 0.000000 [BD 52]  0º
dim_rotation: 0.000000 [BD 50]  0º
 handle stream: +18 @161.5 MISSING (@164.0  @166.0)
dimstyle: (5.1.27) abs:27 [H 0] => DIMSTYLE Standard
block: (5.1.BB) abs:BB [H 0]
 object_map{B9} = 65
 padding: 48/8 (6 bits)
```

DIMBLK1 and DIMBLK2 are defined in EED:
```
EED[2] code: 70 [RC], short: 343 [RS]
EED[3] code: 5 [RC], entity: 0xC800000000000000 [ulong]
```
and
```
EED[6] code: 70 [RC], short: 344 [RS]
EED[7] code: 5 [RC], entity: 0xC800000000000000 [ulong]
```
0xC800000000000000 mean HANDLE 0.1.C8 → _Dot block

DIMSAH is defined in EED:
```
EED[4] code: 70 [RC], short: 173 [RS]
EED[5] code: 70 [RC], short: 1 [RS]
```

There is an error in preview decoding.

michal-josef-spacek, 2023-03-17:
Decoding of Oblique DWGs is with errors.

rurban, 2023-03-18:
And the eed entity ulong's need to be byteswapped to get the handle

timoria21, 2023-03-21:
Thank you for the tests, now it's clear to me how to get these information.

But how are you generating that log exactly? For some reasons I can't make it work on my end.

This is what I'm doing (powershell):
`...\libredwg-0.12.5.5186-win64\dwgread.exe' -v9 -o .\ah_dot_dot.json .\ah_dot_dot.dwg 2>&1 > log.txt`

But the arrowhead handle is still 0 and overall I see a different output.

[image]

[log.txt](https://github.com/LibreDWG/libredwg/files/11027943/log.txt)

michal-josef-spacek, 2023-03-21:
> But how are you generating that log exactly? For some reasons I can't make it work on my end.

In Linux:
```
dwgread -v9 ah_dot_dot.dwg |& less
```

Output:
```
...
handle: 0.1.B9 [H 5] @7.2
EED[0] size: 49 [BS] @8.4
EED[0] handle: 5.1.12 @10.4
EED[0] raw: 49
00060044005300540059004C00450002004657010500000000000000C846AD004601004658010500000000000000C80201
      D   S   T   Y   L   E       F W                     F     F     F X                         
EED[0] code: 0 [RC], wstring: len=6 [RS] "DSTYLE" [TU] @25.4
EED[1] code: 2 [RC], open: 0x0 [RC] @27.4
EED[2] code: 70 [RC], short: 343 [RS] @30.4
EED[3] code: 5 [RC], entity: 0xC800000000000000 [ulong] @39.4
EED[4] code: 70 [RC], short: 173 [RS] @42.4
EED[5] code: 70 [RC], short: 1 [RS] @45.4
EED[6] code: 70 [RC], short: 344 [RS] @48.4
EED[7] code: 5 [RC], entity: 0xC800000000000000 [ulong] @57.4
EED[8] code: 2 [RC], close: 0x1 [RC] @59.4
 @59.4
EED[9] size: 0 [BS] (end)
num_eed: 9
...
```

michal-josef-spacek, 2023-03-26:
Files in this issue:
* ah_dot_arrow.dwg - AC1021 (r2007)
* ah_dot_arrow.json - AC1021
* ah_dot_dot.dwg - AC1021 (r2007)
* ah_dot_dot.json - AC1021
* ah_dot_oblique.dwg - AC1021 (r2007)
* ah_dot_oblique.json - AC1021
* ah_oblique_arrow.dwg - AC1021 (r2007)
* ah_oblique_arrow.json - AC1021
* ah_oblique_oblique.dwg - AC1021 (r2007)
* ah_oblique_oblique.json - AC1021