LibreDWG/libredwg #666: To DXF is not converting all informations.

leshasoft, 2023-03-20:
Hi, I have attached three files. one original DWG file and two dxf files one created using libreDWG and other using AutoCAD. The  file created  using libredwg is not having all information, some information are missing.

[test.zip](https://github.com/LibreDWG/libredwg/files/11014459/test.zip)

rurban, 2023-03-20:
Which esp.? AME and AME_SOL is not yet supported.

leshasoft, 2023-03-20:
EED AME and AME_SOL are not yet supported - this is understandable.
Because of this, the BLOCK does not contain a ENTITIES LINE ?

michal-josef-spacek, 2023-03-20:
There is no support for writing EED in preR13 to DXF

btw: What is AME and AME_SOL?

rurban, 2023-03-20:
I'll check the missing lines in DXF. For DXF there are still some open tickets, esp. preR13 and the ordering.

AME was some early advanced modeling extension ARX product.

michal-josef-spacek, 2023-03-26:
Files in this issue:
* SOLBOX.DWG - AC1009 (r11)
* from_autocad.dxf - AC1009
* from_lidredwg.dxf - AC1009

rurban, 2023-03-26:
I believe this should be fixed now.

michal-josef-spacek, 2023-04-01:
This issue is still here.