LibreDWG/libredwg #677: [BUG] two bugs in dwg2dxf

kdsjZh, 2023-03-30:
Hello, I was testing my fuzzer and found two bugs in dwg2dxf, which is different from #368 and #484

## environment
ubuntu 20.04, GCC 9.4.0, libredwg latest commit 77a85621de6e2c7ea44bc48c02e52dac97778ecb

compile with 
```
./autogen.sh && ./configure --disable-shared && make -j$(nproc)
```
## BUG1
```
/libredwg/programs/dwg2dxf /pocs/pocs/dwg2dxf/poc0.bit_calc_CRC
Reading DWG file /pocs/pocs/dwg2dxf/poc0.bit_calc_CRC
ERROR: Header CRC mismatch 0003 <=> 3C16
Warning:
@0x3d7 => start 0x3c5
ERROR: DWG_SENTINEL_R11_ENTITIES_BEGIN mismatch
ERROR: DWG_SENTINEL_R11_ENTITIES_END mismatch
....
AddressSanitizer:DEADLYSIGNAL
=================================================================
==78==ERROR: AddressSanitizer: SEGV on unknown address 0x62100a3d71b3 (pc 0x563f38494cff bp 0x7ffd51ea1fa8 sp 0x7ffd51ea09e0 T0)
==78==The signal is caused by a READ memory access.
    #0 0x563f38494cfe in bit_calc_CRC /libredwg/src/bits.c:3456
    #1 0x563f38db6295 in decode_preR13_auxheader /libredwg/src/decode.c:6278
    #2 0x563f38e0736a in decode_preR13 /libredwg/src/decode_r11.c:786
    #3 0x563f38d8532a in dwg_decode /libredwg/src/decode.c:217
    #4 0x563f38472a67 in dwg_read_file /libredwg/src/dwg.c:261
    #5 0x563f38470c9c in main /libredwg/programs/dwg2dxf.c:261
    #6 0x7efdc3332082 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x24082)
    #7 0x563f3847153d in _start (/libredwg/programs/dwg2dxf+0x23a53d)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /libredwg/src/bits.c:3456 in bit_calc_CRC
==78==ABORTING
```

## BUG2

```
/libredwg/programs/dwg2dxf -y /pocs/pocs/dwg2dxf/poc1.bit_write_TF
Reading DWG file /pocs/pocs/dwg2dxf/poc1.bit_write_TF
Warning: Invalid sections: 0 != numgaps: 0 + numsections: 26
ERROR: Invalid .props x 33456

ERROR: bit_read_RC buffer overflow at 1704.1 + 0 > 1704
Warning: Unstable Class entity 504 WIPEOUT (0x87f) 111/0
Warning: Unstable Class object 506 MATERIAL (0x481) 114/0
Warning: Unstable Class object 506 MATERIAL (0x481) 122/0
Warning: Unstable Class object 506 MATERIAL (0x481) 130/0
Warning: Unstable Class object 508 TABLESTYLE (0xfff) 164/0
ERROR: Invalid bitsize 445692418 > 1536
...
=================================================================
==83==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x62500000228f at pc 0x5567bafe7bf1 bp 0x7ffc90495880 sp 0x7ffc90495870
READ of size 1 at 0x62500000228f thread T0
    #0 0x5567bafe7bf0 in bit_write_TF /libredwg/src/bits.c:1675
    #1 0x5567bbb416b4 in dwg_convert_SAB_to_SAT1 /libredwg/src/out_dxf.c:2150
    #2 0x5567bbbb2164 in dxf_3dsolid /libredwg/src/out_dxf.c:2398
    #3 0x5567bbbb2164 in dwg_dxf_REGION_private /libredwg/src/dwg.spec:2990
    #4 0x5567bbc6acbe in dwg_dxf__3DSOLID /libredwg/src/dwg.spec:2994
    #5 0x5567bbc6acbe in dwg_dxf_object /libredwg/src/out_dxf.c:2774
    #6 0x5567bbc9ab52 in dxf_entities_write /libredwg/src/out_dxf.c:3635
    #7 0x5567bbc9ab52 in dwg_write_dxf /libredwg/src/out_dxf.c:3796
    #8 0x5567bafc5ce8 in main /libredwg/programs/dwg2dxf.c:339
    #9 0x7f5740a12082 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x24082)
    #10 0x5567bafc653d in _start (/libredwg/programs/dwg2dxf+0x23a53d)

0x62500000228f is located 0 bytes to the right of 8591-byte region [0x625000000100,0x62500000228f)
allocated by thread T0 here:
    #0 0x7f5740e3ca06 in __interceptor_calloc ../../../../src/libsanitizer/asan/asan_malloc_linux.cc:153
    #1 0x5567bb00fa85 in decode_3dsolid /libredwg/src/dwg.spec:2706
    #2 0x21c6  (<unknown module>)

SUMMARY: AddressSanitizer: heap-buffer-overflow /libredwg/src/bits.c:1675 in bit_write_TF
Shadow bytes around the buggy address:
  0x0c4a7fff8400: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c4a7fff8410: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c4a7fff8420: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c4a7fff8430: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c4a7fff8440: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x0c4a7fff8450: 00[07]fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c4a7fff8460: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c4a7fff8470: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c4a7fff8480: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c4a7fff8490: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c4a7fff84a0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==83==ABORTING
```

## POC

[poc.zip](https://github.com/LibreDWG/libredwg/files/11107648/poc.zip)

## Credit

Han Zheng (NCNIPC of China, Hexhive)

rurban, 2023-03-30:
The first is in decode already (dwgread)

kdsjZh, 2023-03-31:
Hello, I just found that the fix for BUG1 seems to be incomplete, here is another poc that can overflow in the latest commit 9df4ec3c524795a401acf371c8a953f5b1a6fadf

```
=================================================================
==19745==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x622000001629 at pc 0x56482d210781 bp 0x7ffe9bda9780 sp 0x7ffe9bda9770
READ of size 1 at 0x622000001629 thread T0
    #0 0x56482d210780 in bit_calc_CRC /libredwg/src/bits.c:3533
    #1 0x56482db3382a in decode_preR13_auxheader /libredwg/src/decode.c:6332
    #2 0x56482db83c5a in decode_preR13 /libredwg/src/decode_r11.c:786
    #3 0x56482db0175a in dwg_decode /libredwg/src/decode.c:217
    #4 0x56482d1eda67 in dwg_read_file /libredwg/src/dwg.c:261
    #5 0x56482d1ebc9c in main /libredwg/programs/dwg2dxf.c:261
    #6 0x7fdf78396082 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x24082)
    #7 0x56482d1ec53d in _start (/validate/dwg2dxf/dwg2dxf+0x23a53d)

0x622000001629 is located 0 bytes to the right of 5417-byte region [0x622000000100,0x622000001629)
allocated by thread T0 here:
    #0 0x7fdf787c0a06 in __interceptor_calloc ../../../../src/libsanitizer/asan/asan_malloc_linux.cc:153
    #1 0x56482d1ecb66 in dat_read_file /libredwg/src/dwg.c:95
    #2 0x1527  (<unknown module>)

SUMMARY: AddressSanitizer: heap-buffer-overflow /libredwg/src/bits.c:3533 in bit_calc_CRC
Shadow bytes around the buggy address:
  0x0c447fff8270: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c447fff8280: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c447fff8290: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c447fff82a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c447fff82b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x0c447fff82c0: 00 00 00 00 00[01]fa fa fa fa fa fa fa fa fa fa
  0x0c447fff82d0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c447fff82e0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c447fff82f0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c447fff8300: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c447fff8310: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==19745==ABORTING
```
[poc0.zip](https://github.com/LibreDWG/libredwg/files/11118093/poc0.zip)

closed 2023-04-04 by rurban