LibreDWG/libredwg #683: Issue in text string conversions in preR13

michal-josef-spacek, 2023-04-01:
When we convert DWG file to JSON and back via libredwg there is difference in text string.

Example:
[second.dwg.gz](https://github.com/LibreDWG/libredwg/files/11128606/second.dwg.gz)
[first.json.gz](https://github.com/LibreDWG/libredwg/files/11128607/first.json.gz)
[first.dwg.gz](https://github.com/LibreDWG/libredwg/files/11128608/first.dwg.gz)

Process:
* Convert first.dwg to first.json
* Convert first.json to second.dwg

All seems ok in dwgread:
first.dwg
```
Add entity TEXT [8] Decode entity TEXT
===========================
Entity number: 8, Type: 7, Addr: 203
layer: 0x1 [H(RSx) 8]
ins_pt: (1.000000, 2.000000) [2RD 10]
height: 2.375879 [RD 40]
oblique_angle: 0.785398 [RD 51]
text_value: "Foo" [TV 1]
```

second.dwg
```
Add entity TEXT [8] Decode entity TEXT
===========================
Entity number: 8, Type: 7, Addr: 203
layer: 0x1 [H(RSx) 8]
ins_pt: (1.000000, 2.000000) [2RD 10]
height: 2.375879 [RD 40]
oblique_angle: 0.785398 [RD 51]
text_value: "Foo" [TV 1]
```

But size of text_value is different. In case of first.dwg is 3, in case of second.dwg is 4 (with \0 string).

As i understand issue is in in_json.c, where json string si converted to string with \0 string and bit_write_TV save this \0 string.

rurban, 2023-04-13:
Because we have special bit_write_TV logic to write that extra \0 for preR2000

rurban, 2023-04-14:
Changed

closed 2023-04-14 by rurban