LibreDWG/libredwg #686: Wrong encoding of handles from JSON to DWG in preR13

michal-josef-spacek, 2023-04-02:
Examples:
[test.json.gz](https://github.com/LibreDWG/libredwg/files/11133369/test.json.gz)
[test.dwg.gz](https://github.com/LibreDWG/libredwg/files/11133370/test.dwg.gz)

I created test.dwg file with CLAYER = 2 (handle r11_idx = 2) and one entity (LINE).
I created test.json from this file via dwgread.
Output:
```.
...
"CLAYER": [0, 0, 0, 0, 2],
...
    {
      "entity": "LINE",
      "index": 8,
      "type": 1,
      "size": 36,
      "layer": [0, 0, 0, 0, 0],
      "start": [ 0.0, 0.0 ],
      "end": [ 100.0, 100.0 ]
    }
...
````
When I want convert this json file to dwg, there is issue with handles.
src/in_json.c contain looking for handle via dwg_add_handleref()
https://github.com/LibreDWG/libredwg/blob/master/src/dwg.c#L2038
But this function not contain r11_idx, then badly match existing handle.
Code ```"layer": [0, 0, 0, 0, 0],``` select same handle as "CLAYER": [0, 0, 0, 0, 2] and rewrite r11_idx to it.
In output DWG document will be last entity layer r11_idx.

I don't know how to fix right, probably we need add r11_idx to dwg_add_handleref() or create another one function.
btw: dwgrewrite is working right

rurban, 2023-04-02:
Yes, we don't handle the new r11_idx at all yet

rurban, 2023-04-04:
We should add the r11_idx as 6th element to the array, and when writing to preR13 use that instead.

michal-josef-spacek, 2023-04-04:
@rurban 
As I understand r11_idx is 5th element of array.
See: https://github.com/LibreDWG/libredwg/blob/master/src/out_json.c#L135

We need to update dwg_add_handleref() with adding r11_idx somehow. But it's public api.

rurban, 2023-04-06:
I've fixed injson, just the absref resolver from r11 is missing. 
if needed at all, but probably we'd need it for outdxf

closed 2023-04-06 by rurban