LibreDWG/libredwg #687: Trunkated floating value in DXF

leshasoft, 2023-04-04:
When outputting to DXF, floating point values are truncated to 18 characters, which distorts the values of the header variables.
I suggest increasing DXF_FLT_MAXLEN.

rurban, 2023-04-04:
Nope, we had at the beginning.

leshasoft, 2023-04-04:
There are more characters in the file created in AutoCAD.

From AutoCAD
```
  9
$PEXTMIN
 10
1.000000000000000E+20
 20
1.000000000000000E+20
 30
1.000000000000000E+20
  9
$PEXTMAX
 10
-1.000000000000000E+20
 20
-1.000000000000000E+20
 30
-1.000000000000000E+20
```
From LibreDWG
```
  9
$PEXTMIN
 10
100000000000000000
 20
100000000000000000
 30
100000000000000000
  9
$PEXTMAX
 10
-100000000000000000
 20
-100000000000000000
 30
-100000000000000000

```

rurban, 2023-04-04:
See, we actually have less with the same precision

leshasoft, 2023-04-05:
1.0E+17 not equal 1.0E+20

In JSON stored value 1.0E+20
 
```
    "PEXTMIN": [ 100000000000000000000.0, 100000000000000000000.0, 100000000000000000000.0 ],
    "PEXTMAX": [ -100000000000000000000.0, -100000000000000000000.0, -100000000000000000000.0 ],
```

michal-josef-spacek, 2023-04-12:
Main issue is limit of DXF RD length defined by:
```
#define DXF_FLT_MAXLEN 18
```

For this case must be value of DXF_FLT_MAXLEN min. 21

rurban, 2023-04-18:
Fixed

closed 2023-04-18 by rurban