LibreDWG/libredwg #692: AddressSanitizer: heap-buffer-overflow on bit_read_T32 libredwg/src/bits.c:2248

fmunozs, 2023-04-09:
Hello,

I am attaching a new file that causes the following segfault with ASAN enabled on dwgread:

[0.zip](https://github.com/LibreDWG/libredwg/files/11186172/0.zip)

This was tested on x86 Debian 11 vm.

```
=================================================================
==1738==ERROR: AddressSanitizer: heap-buffer-overflow on address 0xb09014d0 at pc 0xb301276d bp 0xbf9e3b08 sp 0xbf9e3afc
WRITE of size 2 at 0xb09014d0 thread T0
    #0 0xb301276c in bit_read_T32 /home/user/gcc/libredwg/src/bits.c:2248
    #1 0xb34cbf66 in filedeplist_private /home/user/gcc/libredwg/src/filedeplist.spec:29
    #2 0xb34cbf66 in read_2004_section_filedeplist /home/user/gcc/libredwg/src/decode.c:2759
    #3 0xb34cbf66 in decode_R2004 /home/user/gcc/libredwg/src/decode.c:3414
    #4 0xb34d2c31 in dwg_decode /home/user/gcc/libredwg/src/decode.c:231
    #5 0xb3544642 in dwg_read_file /home/user/gcc/libredwg/src/dwg.c:261
    #6 0x49b71e in main /home/user/gcc/libredwg/programs/dwgread.c:256
    #7 0xb77fee45 in __libc_start_main ../csu/libc-start.c:308
    #8 0x49cf00 in _start (/home/user/gcc/libredwg/dwgread+0x4f00)

0xb09014d1 is located 0 bytes to the right of 1-byte region [0xb09014d0,0xb09014d1)
allocated by thread T0 here:
    #0 0xb7a794bb in __interceptor_malloc ../../../../src/libsanitizer/asan/asan_malloc_linux.cpp:145
    #1 0xb3012655 in bit_read_T32 /home/user/gcc/libredwg/src/bits.c:2240

SUMMARY: AddressSanitizer: heap-buffer-overflow /home/user/gcc/libredwg/src/bits.c:2248 in bit_read_T32
Shadow bytes around the buggy address:
  0x36120240: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x36120250: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x36120260: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x36120270: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x36120280: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
=>0x36120290: fa fa fa fa fa fa fa fa fa fa[01]fa fa fa 02 fa
  0x361202a0: fa fa 00 00 fa fa 02 fa fa fa 02 fa fa fa 02 fa
  0x361202b0: fa fa 02 fa fa fa 02 fa fa fa 02 fa fa fa 00 04
  0x361202c0: fa fa 00 06 fa fa fd fd fa fa 00 02 fa fa fd fd
  0x361202d0: fa fa 00 00 fa fa 00 06 fa fa fd fd fa fa 00 07
  0x361202e0: fa fa fd fd fa fa 00 03 fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==1738==ABORTING

```

rurban, 2023-04-09:
Seems to be already fixed in master.
Which git commit / version did you check?

fmunozs, 2023-04-09:
Hello, I cloned the repository today and  I compiled it with GCC/AFL as follows:

```
git clone  --recurse-submodules https://github.com/LibreDWG/libredwg
cd libredwg
AFL_USE_ASAN=1 ASAN_OPTIONS=detect_leaks=0 CC=afl-gcc CXX=afl-g++ cmake .
AFL_USE_ASAN=1 ASAN_OPTIONS=detect_leaks=0 CC=afl-gcc CXX=afl-g++ make -j
```

Is there any additional information that I can provide?

```
user@debian:~/gcc/libredwg$ git log --name-status HEAD^..HEAD
commit 200078e20173bdc43b096083c01906b3edf5c011 (HEAD -> master, tag: 0.12.5.5340, origin/master, origin/HEAD)
Author: Reini Urban <rurban@cpan.org>
Date:   Sun Apr 9 11:09:54 2023 +0200

    bit_TV_to_utf8: handle out of memory

M       src/bits.c
```

I am compiling with clang/AFL and with plain gcc to see if there is any different behaviour.

rurban, 2023-04-09:
Interesting. I'll check again.

fmunozs, 2023-04-09:
I did some additional debugging @rurban 

I believe there is an overflow check that its failing due to a signed/unsigned addition. 

https://github.com/LibreDWG/libredwg/blob/master/src/bits.c#L2233

For one of the records in this file, **size** becomes 4294967295 and when its added to **dat->byte** (706) it overflows and becomes 705 which is less than dat->size (1414)

```
Breakpoint 1, bit_read_T32 (dat=0xf) at src/bits.c:2224
2224    {
(gdb) n
2227      size = bit_read_RL (dat);
(gdb) print size
$40 = <optimized out>
(gdb) print dat
$41 = (Bit_Chain * restrict) 0xbfffe2d0
(gdb) print *dat
$42 = {chain = 0x8271af0 "\003", size = 1414, byte = 702, bit = 0 '\000', opts = 0 '\000', version = R_2013, from_version = R_2013, fh = 0x0, codepage = 0}
(gdb) n
2229      if (IS_FROM_TU (dat))
(gdb) n
2233          if (dat->byte + size >= dat->size)
(gdb) print size
$43 = 4294967295
(gdb) print dat->byte
$44 = 706
(gdb) print dat->byte  + size
$45 = 705
(gdb) print dat->size
$46 = 1414
```

rurban, 2023-04-12:
Should be fixed now

closed 2023-04-12 by rurban